efornara/frt #1: cross compiling

zoiba, 2017-05-29:
well, is not really an issue: have you any advice for cross compiling from Ubuntu 16.04?
tried your binary on Raspberry 3 and it works quite fine, I wish to try to compile it on my own, tried to cross compiling on Ubuntu machine with no luck, is there a better way?

efornara, 2017-05-29:
I build the executables on a PC running debian stable (jessie). I guess that same procedure should work on Ubuntu 16.04 too, but I am not sure if that is what you are looking for: I don't use the cross compiler toolchain, just the qemu-arm-static trick.

On my old and underpowered dual core, this:

    time nice scons platform=frt target=release tools=no frt_arch=pi2 -j 3

takes around two hours from scratch. I didn't bother setting up the cross compiler toolchain because I am not familiar with scons and so far I have managed without: I leave all the generated .o files in place, and I rarely trigger a full recompile.

For the qemu-arm-static trick, see for example here:

https://wiki.debian.org/RaspberryPi/qemu-user-static

I don't actually resize the image, but I just copy its content to a normal directory, like suggested here ("sudo cp -r /mnt $SYSROOT" is the relevant bit):

https://github.com/davidsiaw/SDL2/blob/master/README-raspberrypi.txt

zoiba, 2017-06-01:
thanks for the tip, I will try ASAP

closed 2018-03-28 by efornara