efornara/frt #2: Not receiving non-keyboard input

JanGross, 2017-11-12:
I'm trying to run a game using non-keyboard/mouse inputs. But hitting the buttons does not trigger any input event at all. It works fine on other platforms.

efornara, 2017-11-12:
I am not sure what you mean by "non-keyboard/mouse inputs" / "buttons". Do you mean mouse buttons, joystick buttons, or something else?

Could you post the relevant input mapping and maybe a snipped of code where you show how you handle the event?

Joysticks are not supported yet. It should be fairly easy to import and integrate the official linux driver from godot, but it hasn't been done yet.

JanGross, 2017-11-13:
I do mean Joystick buttons. My input config looks like this:  
```
[input]

Start=[jbutton(0, 6)]
B=[jbutton(0, 8)]
Coin=[jbutton(0, 7)]
A=[jbutton(0, 9)]
Coin2=[jbutton(0, 4)]
Select=[jbutton(0, 5)]
Y=[jbutton(0, 10)]
X=[jbutton(0, 11)]
Blin1=[jbutton(0, 0)]
Blin2=[jbutton(0, 2)]
J_up=[jaxis(0, 2)]
J_down=[jaxis(0, 3)]
J_left=[jaxis(0, 0)]
J_right=[jaxis(0, 1)]
```
I looked around the frt and godot source but have a hard time figuring out how stuff works.

Faless, 2017-11-13:
@JanGross I made a small patch to get joypad support on the PI here https://github.com/Faless/frt/tree/joyrebase
I tested it with Godot 2.1 and it works (although it requires a small patch to get the proper mapping):
```
diff --git a/main/input_default.cpp b/main/input_default.cpp
index 5e923af9f..4799def2e 100644
--- a/main/input_default.cpp
+++ b/main/input_default.cpp
@@ -620,7 +620,7 @@ static const char *s_ControllerMappings[] = {
        "d814000000000000cecf000000000000,MC Cthulhu,leftx:,lefty:,rightx:,righty:,lefttrigger:b6,a:b1,b:b2,y:b3,x:b0,start:b9,back:b8,leftshoulder:b4,rightshoulder:b5,dpup:h0.1,dpleft:h0.8,dpdown:h0.4,dpright:h0.2,righttrigger:b7,",
 #endif

-#if X11_ENABLED
+#if X11_ENABLED || FRT_ENABLED
        "0000000058626f782047616d65706100,Xbox Gamepad (userspace driver),a:b0,b:b1,x:b2,y:b3,start:b7,back:b6,guide:b8,dpup:h0.1,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,leftshoulder:b4,rightshoulder:b5,lefttrigger:a5,righttrigger:a4,leftstick:b9,rightstick:b10,leftx:a0,lefty:a1,rightx:a2,righty:a3,",
        "0300000000f000000300000000010000,RetroUSB.com RetroPad,a:b1,b:b5,x:b0,y:b4,back:b2,start:b3,leftshoulder:b6,rightshoulder:b7,leftx:a0,lefty:a1,",
        "0300000000f00000f100000000010000,RetroUSB.com Super RetroPort,a:b1,b:b5,x:b0,y:b4,back:b2,start:b3,leftshoulder:b6,rightshoulder:b7,leftx:a0,lefty:a1,",
```

If @efornara is interested I can try it out with Godot 3.0 and PR it.
It's a bit hacky since I had to copy the `joystick_linux.h` from the `x11` platform and include `joystick_linux.cpp` from there (which is better than copying it at least).

BTW, great work on this platform, I think we may be able to convince the devs to include your work in godot if you are interested.

efornara, 2017-11-13:
@JanGross I see. I've also just checked your repositories and I guess the project you're working on is this: https://github.com/JanGross/retro_missile. I'll check it out shortly...

Thanks @Faless That's pretty much how I would have done it. That's not hacky at all by frt standards: I did something similar for the x11 key mapping, and it's nothing compared to the big hack that is the godot 2/3 handling in os_frt.cpp  :-) The only thing I need to check is if libudev.so is included in raspbian lite.

As of now, frt doesn't compile against godot 3.0. I've put it on hold and I'm waiting until the godot codebase stabilizes a bit (release candidate or even later). Having a patched input_default.cpp upstream would be really nice, but I'm not sure how a flag for a non-standard platform would be received.

About including frt in godot, The elefant in the room is vc4 support. If/when the foundation makes it the default in raspbian, frt will become far less useful. The only bit worth saving might then be the dl/ code for x11 vs wayland... and the support for some cheap boards based on sunxi (but this s really too niche to merit inclusion into godot). I'd say it's better to wait. 

I'm compiling https://github.com/Faless/frt/tree/joyrebase right now, and if it runs retro_missile I see if I can upload the binary to sourceforge.

efornara, 2017-11-13:
For what I can tell, it seems to work (pi1 version tested on a Pi 2 Model B).

Congratulations and thanks! That's the first godot game that I know of that was written taking the Raspberry Pi into account.

The only (minor) issue I had was that I had to attach a mouse and click on the title to make it disappear.

The pi1 binary should be available now, while the pi2 binary should be available shortly. I'm not planning to release a new version just yet, so you'll find it in the 0.9.2 folder.

Faless, 2017-11-13:
> but I'm not sure how a flag for a non-standard platform would be received.

Maybe instead of checking for `X11_ENABLED || FRT_ENABLED` I can propose to check for `JOYDEV_ENABLED` which is defined in X11 and required for `joystick_linux.cpp` to be included. (or rename JOYDEV_ENABLED to `LINUX_JOYPAD_ENABLED` or even using `#ifdef __linux__` like it's done for android since apparently other UNIXes like BSD won't trigger the inclusion of joysticks support).

I'll try to poke them and see what they think about it.

JanGross, 2017-11-14:
@efornara The project is actually my first Godot game   
and was originally written for Android phones  
([Google Play link](https://play.google.com/store/apps/details?id=com.minzkraut.tbmc))  
hence the resolution and control issues.  

I just made a quick control port to see if it would be possible to make my own games for a custom built Bartop Arcade machine powered by an RPi 3 B.

I will give @Faless's patch a try later today. Thanks for the efforts

JanGross, 2017-11-17:
I just tried the patch with my hardware and it works perfectly, thank you 👍 
Time to make some arcade games 👾

efornara, 2017-11-26:
@Faless Can you create a PR as it is (no need to compile against Godot 3.0), so I can merge it?

Faless, 2017-11-26:
Sure, I'll PR it later today

Faless, 2017-11-26:
> @Faless Can you create a PR as it is (no need to compile against Godot 3.0), so I can merge it?

Done :+1:

efornara, 2017-11-26:
Thanks #3

efornara, 2017-11-26:
Thanks #3

closed 2017-11-26 by efornara