efornara/frt #4: GPIO and Python(uinput)

leofrozen, 2019-01-22:
I'm trying to set some buttons using the "gpiozero" and "uinput" libs from python, but the game never grabs the key emitted. 
**Has any way to my game be controlled by GPIO (with python or others ... )?**

For example:
"
```
from gpiozero import Button
import uinput

button = Button(26)
device = uinput.Device(uinput.KEY_Q, uinput.KEY_W, uinput.KEY_E,)

while True:
    if button.is_pressed:
        device.emit_click(uinput.KEY_Q)

```
"

efornara, 2019-01-31:
frt, when running with the legacy driver, opens /dev/input devices directly, so this might be the reason why you don't receive the events you generate. To confirm that this is the issue, one thing I would try is to enable the new X11 OpenGL driver, start X11 and start the game from there.

If this is the case, and you are writing the game for yourself (so you don't care about asking your users to switch to the new driver too), you could consider enabling the new X11 OpenGL driver permanently, and switching from godot/frt to the standard godot/x11 (the standard godot/x11 has better support for x11 all around).

As for frt, if I had to use GPIO myself, I suppose I would first try to fork and patch frt, but I would not recommend this route to most developers. Another, perhaps cleaner, option would be to try to write a godot module. One problem I can see here, beside the complexity, is permissions issues.

The fact is that I haven't thought about this yet. What you are doing seems easy enough for most developers, so I'll have a look at the trade-offs (e.g. ease of use, compatibility among distros, etc...) of using a higher level library to access input devices.

Any suggestions are appreciated.

hiulit, 2019-07-26:
I don't know if this is related, but I think it is. Here goes:

I'm using [GPIOnext](https://github.com/mholgatem/GPIOnext), a utility that convert GPIO buttons to key strokes. It works really well with my arcade RetroPie setup. I can use the joystick and all the buttons will all the menus and roms, but it's not recognizing the gpio/key strokes when playing a Godot game. It only recognizes an actual keyboard.

efornara, 2019-07-30:
@hiulit Yes, it could be related. It has been a while now, and I expect to evaluate options about GPIO handling in the next few days. I should be able to check that utility too.

hiulit, 2019-07-30:
Thanks a lot! You are making so happy for allowing me play Godot games (and my own game at some point) in my arcade cabinet!

In case you start working on it, check out [Adafruit-Retrogame](https://github.com/adafruit/Adafruit-Retrogame) also because that's the most famous.

On the other hand, shouldn't this utilities work out of the box? I mean, in the end, they just convert GPIO inputs to key strokes, right? So Godot should just "listen" to those key strokes? Why doesn't it work like that?

efornara, 2019-08-11:
I have added support for selecting a custom keyboard device. It seems a good compromise: frt code doesn't become too complex, but it should still be possible to use virtual keyboard libraries without too many hops. It isn't automatic, however.

The daemon generating the virtual events should already be running when frt starts.

In the script starting frt, you can set the environment variable `FRT_KEYBOARD_ID`. If its value starts with a `/`, it is interpreted as the full file path of the device. If it doesn't, it is interpreted as the name of the device. Using the name is recommended. To find it out, type this while your daemon is running:

    cat /proc/bus/input/devices

You should be able to spot which device is the virtual one. I have only tested it with uinput for now. IIRC, the relevant line looked something like this:

    N: Name="python-uinput"

Therefore, a script to start frt could look something like this:

    export FRT_KEYBOARD_ID="python-uinput"
    frt_311_pi1_kbd.bin --main-pack demo.pck

I don't expect to release a new version of frt for another couple of weeks at least (I plan some improvements in the x11 code path), so I would appreciate if anyone wants to have a try in the meantime. I have uploaded a patched version of frt to sourceforge (demos).

@hiulit I had a look at GPIOnext. I am not sure if I will try it, as it looks like it has to be installed as a service and I would prefer not to mess with my raspberry installation. It seems it is using evdev as the library to generate the virtual keyboard events, which I think it is also the library used by Adafruit-Retrogame. My idea is to test evdev shortly (but not GPIOnext and Adafruit-Retrogame specifically).

hiulit, 2019-09-02:
Thanks for taking the the time to look into evdev. Those 2 programs are the ones most people use when using arcade buttons connected to the GPIO of the RPi. I took a look at both projects and they evdev.

BTW, what's the difference between the solution you gave above and using evdev? At the end, what those programs do is create a virtual keyboard, so maybe adding the ID of that keyboard would work as well?

Thank you for your time!

efornara, 2019-09-03:
I think that evdev itself uses uinput, so in theory it should work the same way. Looking at the code, the IDs should be (GPIOnext):

    export FRT_KEYBOARD_ID="GPIOnext Keyboard"

and (Adafruit-Retrogame):

    export FRT_KEYBOARD_ID="retrogame"

But again, that's the theory. I haven't actually tested if they really work.

hiulit, 2019-09-04:
Thanks! I'll try whenever I can. So, as an example, should I use the code like this?

```
export FRT_KEYBOARD_ID="GPIOnext Keyboard" frt_311_pi2.bin --main-pack lalala.pck
```

efornara, 2019-09-04:
Sort of. If you use a single line command, you shouldn't use export:

    FRT_KEYBOARD_ID="GPIOnext Keyboard" frt_311_pi2.bin --main-pack lalala.pck

It this doesn't work, and you are sure that the ID is correct (use `cat /proc/bus/input/devices` to check), as a last resort you can always use an intermediate script:

    /home/pi/bin/lalala.sh

In that case, you can use export (`lalala.sh`):

    #! /bin/sh
    
    export FRT_KEYBOARD_ID="GPIOnext Keyboard"
    exec frt_311_pi2.bin --main-pack lalala.pck

efornara, 2019-10-17:
I've tested it with python / evdev and it works fine for me.

closed 2019-10-17 by efornara

hiulit, 2019-11-06:
I've tested it with a RPI 3B+ using `FRT_KEYBOARD_ID="GPIOnext Keyboard" frt_311_pi2.bin --main-pack lalala.pck` and it works! Thanks :)