efornara/frt #5: Black screen (broken render) in current godot master (3.1)

Faless, 2019-02-28:
Rendering seems to be broken with the FRT platform on RPI3 (raspbian stretch):
Compiled with:
`scons platform=frt target=release frt_arch=pi3 module_gdnative_enabled=no module_webm_enabled=no builtin_freetype=yes -j9`

Tracked back to https://github.com/godotengine/godot/commit/aab8f443f90df21f84dd306600db0c5d1378ade9 (first broken).

Show splash image, then black screen with error:
```
WARNING: _render_target_allocate: Could not create framebuffer!!
   At: drivers/gles2/rasterizer_storage_gles2.cpp:4259.
```

Faless, 2019-02-28:
The status seems to be:
`GL_FRAMEBUFFER_INCOMPLETE_ATTACHMENT` (36054)

Faless, 2019-02-28:
After further debugging, this only happens with the bcm driver, the VC4 mesa driver works well instead.
I suspect this might be a limitation of the driver itself but my knowledge in those matters is pretty limited, I'm happy with the mesa driver, which also seem to have much better performances

efornara, 2019-03-12:
I haven't been following closely what is happening upstream over the last few days. The pace is just too crazy for me.

Anyhow, I did try the 2d platformer now ( https://github.com/efornara/frt/commit/836edada9e4ea56f4c7c8b3c367b13eeb39cedb9 against https://github.com/godotengine/godot/commit/dba5d2bb278318456bec1f13fb928950eb2a37e4 ), and it works for me.

Maybe the problem has been fixed already, or maybe you were testing a more complex project than the 2d platformer. If you confirm that this is still an issue, I can open an issue in https://github.com/godotengine/godot. I think there are some Android phones that have a VideoCore IV GPU and their drivers is likely to be very similar to the legacy Raspberry Pi driver. See https://github.com/godotengine/godot/issues?utf8=%E2%9C%93&q=is%3Aissue+videocore

Faless, 2019-03-12:
> I haven't been following closely what is happening upstream over the last few days. The pace is just too crazy for me.

Eheh, agreed, there's been quite some changes lately, we should be finally close to release :).

> Maybe the problem has been fixed already

Yeah, that might be possible, there's been some changes to the GLES2 render in the last few days.
I did try with a very simple project (basically an empty project, with just a scene with a panel).
I'll retest ASAP and post my findings here.

Faless, 2019-03-12:
I confirm it's working with current master.
It was indeed a Godot issue, closing as it's now fixed.

closed 2019-03-12 by Faless

WiggleWizard, 2019-03-31:
I'm having this exact issue on an RPi2 currently. An attempt to build Godot has taken about 25 hours but eventually died due to lack of memory.

Any fix for this? Do I just need to export from a newer 3.1 (master branch compiled tools) or does Godot + FRT need recompiling on the RPi?

Faless, 2019-03-31:
> Do I just need to export from a newer 3.1 (master branch compiled tools) or does Godot + FRT need recompiling on the RPi?

You should always export from an editor build from the same sources (commit) you used to built the FRT + Godot binary.

> An attempt to build Godot has taken about 25 hours but eventually died due to lack of memory.

I suggest you try cross compiling. I personally use QEMU to compile from my linux desktop for the RPI.