efornara/frt #6: Material Tester Demo Doesn't Work on Pi

triverske, 2019-03-16:
I was unable to get the material tester demo working on Pi even with --video-driver GLES2, is anyone else able to replicate this issue and is it a frt or a Godot bug?

efornara, 2019-03-18:
To be honest, when I uploaded the demo I had high-end boards in mind, and I've never tried it myself on a Raspberry Pi. It might work, but it wouldn't surprise me if it doesn't. I'll have a go too in the coming days.

It could be the export. If you downloaded it from sourceforge, the problem might be that you are using an export for Godot 3.0. I should be able to upload new exports in the coming days. If, in the meantime, you want to try it yourself, you can export a ZIP or a PCK for Android (safer option) and use that.

It could be a driver issue. It might work on vc4 and not on bcm. You could try to enable (disable) the new driver and see if that solves the problem. If the demo doesn't work on the vc4 driver, it would make sense to test it on a standard Godot build to to see if it really is a FRT bug. If it isn't, it is better to open an issue upstream.

triverske, 2019-03-18:
Further testing shows it being unable to run on my odroid xu4. 

I exported the material testers demo from my own copy of 3.1 stable

efornara, 2019-03-18:
Since I've never received any feedback on other boards, would you mind giving me some feedback on your odroid xu4?

By "it", do you mean the material tester demo or FRT in general? Can you run the 2d platformer demo on your odroid xu4? If not, can I ask you which distro are you using, and to post how a program that does work (es. es2gears) is linked? For example, the result of something like this:

    $ which es2gears
     /usr/bin/es2gears
    $ ldd /usr/bin/esgears
    ???

triverske, 2019-03-18:
2D stuff appears to work fine, including 2D platformer and one of my own test projects on both odroid and pi. On odroid the gles3 backend is working for this as well

3D seems to be causing problems on both backends

triverske, 2019-03-19:
Exporting the android pck fixed a lot of issues and I was able to get it running. Lots of driver bugs with failed shaders and whatnot but I got a window to open.

efornara, 2019-03-23:
I've just had a quick try, and I personally cannot even start the demo. I had a texture format issue (as I was kind of expecting), but I also had a precision related issue in the shader.

I should be able investigate a bit more soon...

triverske, 2019-03-23:
Yeah I have issues loading textures as well, I've tried a couple different things and on occasion I can get one or two of the material balls to load, but I'm not sure what the cause is.

Faless, 2019-05-18:
> but I also had a precision related issue in the shader.

@efornara it seems RPI needs high precision by default like WebGL in `drivers/gles2/shader_gles2.cpp`.
```
#ifdef JAVASCRIPT_ENABLED
	strings.push_back("#define USE_HIGHP_PRECISION\n");
#endif
```

Removing the `ifdef` (enabling high precision) fixed shaders on the PI.
Not sure how this should be addressed in the engine.

efornara, 2019-05-19:
This is going to be a long post, but I suppose it is overdue.

@Faless I did try commenting out these ifdefs a while back to force high precision, but in my case all I got was a kernel panic (legacy driver). I hope this was due to GPU resources using too much memory and exposing corner cases in the whole Pi graphics stack.

Note that I get the same precision error on frt running on an Intel PC, so there is a small but non insignificant chance that the "angle does not like" comment upstream might turn out to be optimistic and some android devices might have the same problem too. I should probably open an issue upstream to call for feedback on that: if it can be shown that some android phones have the same issue on a stock godot, upstream is likely to take it much more seriously. This would also help me not to fork the engine (I doubt I can do some hacks on the platform).

By the way, forcing high precision on Intel PC does let the shaders compile and does not crash the machine, but the resulting rendering is broken.

To be honest, I am not that interested in chasing bugs in the Material Tester Demo, just to be greeted with a glorified slideshow if everything goes well. I just don't think this is the right kind of demo for the Raspberry Pi, and for frt in general.

I did try (admittedly, just a quick test) to come up with a minimal demo (an opaque plane and a semi-transparent cube), but I remember being puzzled when inspecting the exported textures in the PCK. I might try again, but it would be just me scratching an itch; a spinning cube working would be nice, but:

What I am looking forward to is the upcoming demo targeting low-end devices. See here: https://twitter.com/reduzio/status/1110400672258678784 I have already mentioned it here: https://github.com/godotengine/godot/issues/27407

To sum up:

1. I expect to do a serious push to try to get 3D working once a preview of the new demo is released.

2. If the problem affects android too, we might be lucky and get the problem fixed upstream.

3. I might have another go with a minimal demo, but don't count too much on this.

efornara, 2019-09-23:
Just a quick update. No good news, I am afraid.

I have tried godot 3.2-dev not long ago. USE_HIGHP_PRECISION seems to be fixed now, however I still get a kernel panic on the legacy driver. I think that with one Environment setting it didn't crash, but it still got nothing renderer (IIRC, something like the memory was exceeded, so the legacy shader compiler might not be able handle the complexity of the 3D shaders produced by godot 3.x). Maybe it is my version of the firmware, and some others might have no problems. As with 3.1, no problems with 2D on 3.2-dev.

The new driver (X11) was fine, at least for the simple scene that I have tested, but, there is an upstream issue with ETC/S3TC textures on Android that, unsurprisingly,  also affects frt (https://github.com/godotengine/godot/issues/28308). You can work around it, but it is a pain.

There is still time until 3.2 is released, but probably my advice would be to use the upstream X11 platform if you do 3D. It is not ideal either. You are likely to export S3TC textures without noticing, and get the worst of both worlds: compression artefacts on one side and no GPU memory saving on the other: since the hardware doesn't support S3TC, the desktop OpenGL driver will likely uncompress your textures behind the scenes. Still, it is better than publishing a game and having it crash the machines of users attempting to play it in on the legacy driver.

efornara, 2021-01-11:
Closing this old issue. It might make more sense to focus on single games. See https://github.com/efornara/frt/issues/27

closed 2021-01-11 by efornara