efornara/frt #7: Error: CanvasShaderGLES2

hiulit, 2019-03-25:
**Godot version:**
3.1-stable

**OS/device including version:**
Debian (Raspbian) Stretch / Raspberry Pi 3 Model B

**Issue description:**
I have a `GLES2` project with no shaders (that I'm aware of). The only "special" thing it's some `CPUParticles2D`.
I exported the project using `frt_094_310_pi2.bin` and when I launch it, these errors appear (and the `CPUParticles2D` don't show up)

```
ERROR: _display_error_with_code: CanvasShaderGLES2: Program linking failed:
ERROR:LINK-4 (link time, line -1) Too many attribute values
    At: drivers/gles2/shader_gles2.cpp:129.
ERROR: get_current_version: Method/Function Failed, returning: _null
    At: drivers/gles2/shader_gles2.cpp:414.
ERROR: bind: Condition ' fversion' is true. returned: false
    At: drivers/gles2/shader_gles2.cpp:88.
ERROR: _get_uniform: Condition ' fversion' is true. returned -1
    At: drivers/gles2/shader_gles2.h:254.
```

**Steps to reproduce:**
Open the `test-rpi.zip` below in a Raspberry Pi 3 Model B and run the following command:

```
cd test-rpi/
sudo chmod +x frt_094_310_pi2.bin # To give permissions to the file, if needed.
./frt_094_310_pi2.bin
```

I've also added the source code of the project, just in case it could be needed.

**Minimal reproduction project:**

[test-rpi.zip](https://github.com/godotengine/godot/files/3003891/test-rpi.zip)
[project-source-code.zip](https://github.com/godotengine/godot/files/3003903/project-source-code.zip)

efornara, 2019-03-25:
See upstream: https://github.com/godotengine/godot/issues/27407

closed 2019-03-25 by efornara

Jose-Moreno, 2020-06-24:
@efornara Hi. First of all excellent work with FRT. Apologies for necroing this issue, however I wanted to ask something related to this problem. 

I'm currently trying out the **frt_097_320** binary for the Pi2. I'm also using GLES2, however I don't have a single shader (not that I can tell) nor do I have CPUParticles. I'm only using control nodes, animated sprites and animation players, however I get the exact same error as hiulit regarding GLES2, and due to this FRT's screen starts flickering and  I can only get out using Win + Q.

I compiled the game following some of your advice on the same thread the two of you discussed 1 year ago on reddit. I'm using the X11 Template, but using Windows 10, would I need a linux box instead?. Is there any way I can make this work with FRT? Also I intend to target the RPi4, but I don't have access to one yet, only to the Pi2, would that also make a difference?

I've tried the demos you provide on sourceforge and they work just fine, so I don't understand why my poor test app doesn't work haha 😅 

FYI I've previously tried to compile Godot from source on the Pi2 and I was anything but sucessful. After 8 hours of retrying the compilation after LLVM ran out of memory It was simply impossible. Honestly any comment or pointer will be appreciated beyond measure. Thanks in advance 🙇

efornara, 2020-07-13:
Sorry that I didn't see you comment before. I usually only notice open issues.

I don't know about the GLES2 error you are having. As far as I know, plain 2D sprites should be fine. If you are still stuck, you can post more info here (log and possibly a minimal project), but more likely than not I will have to refer you upstream.

To export a game with the editor, Win 10 is fine. Linux (on ARM is easier) is only needed to generate the export templates yourself, either X11 or FRT. As for Pi4, there might be corner cases that I am not aware of, but in general testing on a Pi2 should be fine.

I don't think I have ever tried to compile godot 3.x on a Pi2 (IIRC, only godot 2.1). 8 hours seems too much. My advice would be to attach an external USB drive, put the godot and frt trees in there and add a swap partition (I use a 2G). Not starting the desktop might also help. I think the only case when I have problem with lack of memory is if I try to compile the editor with LTO enabled.

Jose-Moreno, 2020-07-25:
@efornara Thanks for the answer, I apologize for writing so late; i've been swamped with work. In the end I had to switch gears and managed to find a Godot binary of the editor itself compiled specifically under the Pi4 in order to test the app. Fortunately it worked exactly as it did in my development environment on Windows , plus I had no GL related errors like before.

I did however have some engine errors that are unix related (unable to get modified time error), but that's a separate upstream issue that I've already reported.

Either way thanks for taking the time to answer! Keep up the great work 😄