efornara/frt #8: Change/remap keyboard keys recognized by FRT

hiulit, 2019-03-26:
Is there a way to change, for example, Win + Q, for something else?

efornara, 2019-03-26:
Not really.

The idea is, hopefully in the next version of FRT, to disable the meta keys altogether when using X11 (they often interfere with window managers and they are not really needed there) and to allow the user to disable them even when using the legacy driver (most likely, with a custom attribute).

These keys are there mostly as a way to allow the user kill "generic" demos that might not have any other ways to terminate. Some official 2.1 demos were written this way.

The expectation is that anyone writing a game would provide other ways (e.g. an in-game menu) to quit the game.

hiulit, 2019-03-26:
Yeah, you are right. The in-game menu should be fine.

closed 2019-05-19 by efornara