efornara/frt #9: Reproduced problem and general discussion

lukas-durica, 2019-06-07:
You wanted to know if someone reproduced your problem
https://github.com/godotengine/godot/issues/2671#issuecomment-453222752

My Error Report:
`ERROR: ContextGL: Condition ' singleton ' is true.
   At: drivers/gl_context/context_gl.cpp:44.
ERROR: ContextGL: Condition ' singleton ' is true.
   At: drivers/gl_context/context_gl.cpp:44.
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
OpenGL ES 2.0 Renderer: Gallium 0.4 on VC4 V3D 2.1
ERROR: initialize: Directional shadow framebuffer status invalid
   At: drivers/gles2/rasterizer_scene_gles2.cpp:3369.
WARNING: _get: Property not found: physics/2d/thread_model
   At: core/project_settings.cpp:193.
ERROR: PhysicsServer: Condition ' singleton != __null ' is true.
   At: servers/physics_server.cpp:726.`

I used FRT. More specifically frt_095_311 on Raspberry PI 3 B+ (32 bit) with Raspbian. The application was developed under the Windows 10 with Godot 3.1.1.
For now, I don't need GUI (I need only prints in the console). After the error report, my application ran without any problems.

I have a noobish question if you don't mind. I have limited knowledge of Linux. Could you explain the next paragraph?

> Export templates for the official X11 platform can be easily generated and they work on a Raspberry Pi, provided that you switch to the new GL driver. If this limitation is acceptable to you, there is no reason to use FRT. Also, the X11 platform supports the X11 environment far better than FRT.

More specifically:

> Export templates for the official X11 platform can be easily generated

How?

Could you point me to some kind of tutorial?

efornara, 2019-06-07:
@lukas-durica 

> You wanted to know if someone reproduced your problem [...] After the error report, my application ran without any problems.

That request is quite old. I was specifically asking if anyone else was noticing a severe performance degradation in a pre-release version of godot 3.1. That problem has been fixed before the final godot 3.1 was released.

As for the errors you see, I don't think there is much to be done there.  I have seen godot showing different error messages in different platforms, operating systems, graphics drivers... Most of them are harmless.

> How?
>
> Could you point me to some kind of tutorial?

This:

> Export templates for the official X11 platform can be easily generated

Should be qualified like this:

> Export templates for the official X11 platform can be easily generated [if you have some familiarity with Linux and the command line environment]

The easiest way is probably to generate them on the Raspberry Pi itself (running Raspbian).

You can follow the official instruction here:

https://docs.godotengine.org/en/3.1/development/compiling/compiling_for_x11.html

Since Raspbian is derived from debian, you should use the Debian / Ubuntu oneliner, followed by:

    sudo apt-get install clang

To also install the clang compiler. Instead of the official build command:

    scons platform=x11 tools=no target=release bits=32

You should try the line I posted in the comment following the one you linked:

    scons platform=x11 target=release tools=no use_llvm=yes CCFLAGS="-mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -mlittle-endian -munaligned-access" -j 4

As a bonus you can also read the frt instructions here:

https://github.com/efornara/frt/blob/master/doc/Compile.md

They contains some tips.

How successful you are depends mostly on your experience with Linux and the command line environment:
* If the official instructions make sense, it should be easy.
* If the official instructions are puzzling in a couple of places, it should be doable after some googling and some research on your part. If you are stuck, feel free to ask here.
* If the official instructions make no sense at all, I am afraid generating the export templates is not for you. @crabcrabcam used to build and release export templates for the Raspberry Pi. You could try to ping him or ask on social media (e.g. reddit) if anyone else is willing to do it.

closed 2019-06-07 by lukas-durica

lukas-durica, 2019-06-07:
I needed a quick solution. Sorry for bothering you. You can delete the issue.