efornara/frt #11: VideoPlayer .pck/.zip issues

luveti, 2019-09-06:
Does **frt** support video playback? When I go to launch my game it says the webm file doesn't exist in the .pck. But it works fine on my Windows machine.

efornara, 2019-09-06:
The webm codec is disabled in frt. See:

https://github.com/efornara/frt/blob/master/release.sh

IIRC, the underlying problem is that configuration detection upstream uses hard-coded platform names a lot. See for example how webm_cpu_arm is set here:

https://github.com/godotengine/godot/blob/master/modules/webm/libvpx/SCsub

On the official x11 / arm platform (i.e. *not* frt), the webm codec  might work.

On frt, you can try to use the theora codec. I remember testing theora video playback in godot 2.1.x and it worked, so there is a good chance that it should work on godot 3.1.x too. The codec quality is unfortunately likely to be lower. 

Before the next release, I'll try to have a go to see if I can come up with some workaround on my side, but there's no guarantee I'll be able to do it.

luveti, 2019-09-10:
Ahh I see! Thanks for the quick reply! Theora works well but like you said the quality and file size are much worse. Do you happen to know if theora is hardware accelerated in the arm64 build? An otherwise very smooth animation chugged with a moderate size video playing

efornara, 2019-09-11:
I don't think there is any hardware acceleration when playing videos in Godot, regardless of version, architecture or platform. You might get some CPU-specific optimizations (e.g. neon in webm), but, if I remember the engine code correctly, the video is in any case decoded by the CPU and uploaded to the GPU frame by frame as a texture.

Speed-wise, webm might be better in the sense that it has better *CPU* optimization and that might just be enough for your needs. You can try that now by compiling a plain x11 platform build. Or you can wait to see if I can get webm included in the next version of frt. I haven't tried yet. I expect to release the next version of frt 1-3 weeks.

frozenware, 2019-09-28:
Brother, I tried to run webm using Godot 3.1 and an x11 platform, but the video is very slow. The RPI 3b+ CPU always remains above 100%.
What is the "secret" that makes your FRT platform much faster than the x11 platform?

efornara, 2019-09-29:
The FRT platform shouldn't be faster than the x11 platform.

If it is, either 1) the experimental OpenGL driver is not configured, so the FRT build is getting hardware accelerated graphics and the x11 build isn't, or 2) the x11 build has been compiled without CPU-specific optimization options, something like `CCFLAGS="-mcpu=cortex-a7 -mfpu=neon-vfpv4 -mfloat-abi=hard -mlittle-endian -munaligned-access"` added to scons.

Another (remote) possibility might be the compiler: FRT official binaries are generated by an old compiler: g++ 4.9. In theory modern compilers should be better, but non-LTO old g++ might be better than non-LTO new clang++.

Most likely, FRT would get very slow webm videos with CPU above 100% too, assuming that FRT gets webm support in the first place. I had another look at the code a few days ago, and I am not optimistic: I can only think of a very ugly workaround to get webm included without forking the engine. If I also get unusable performance, there is really no point in doing it in the first place.

closed 2022-01-29 by efornara