efornara/frt #12: More of a question than an issue. FRT on RPi4 FBDEV

peicou, 2019-09-20:
I built it for an RPI4 on raspbian lite with the intention of running Godot games directly on the FB so I don't have to deal with the X11 overhead, which I believe is the whole point of your platform.

I'm failing to initialize EGL. envprobe.c checks for vc4 (libbrcmEGL) but finds no X11. Do I need a special driver to run GLES2/3 directly to the FB?

efornara, 2019-09-20:
Sorry, but frt doesn't work without X11 on the RPI4.

The doc reflected the situation in the past where "x11" vs "console" was sort of OK, but now with RPI4 it makes sense to make it more technical. It really is "x11" (Pi 2-4 gui) vs "bcm" (Pi 0-3 console) vs "drm/gbm" (Pi 4 console, I suppose).

The code for the Pi 4 console is simply not there. The fbdev code you see in frt is a first attempt at targeting old chips/drivers (i.e. closed source Mail 400 drivers) and it is not meant to work on more modern chips/drivers like the Pi 4.

Assuming that the Pi 4 works as I expect (i.e. using drm/gbm actually works). it should be technically possible to write a frt module for the Pi 4 console. However, to be honest, I am not planning to write one right now (besides, I still don't have a Pi 4, so I couldn't test it). Realistically, you are looking at 2-3 months at least. If anyone listening wants to have a go before me, patches are welcome.

peicou, 2019-09-22:
Thank you so much! You are doing (did?) amazing work and I'll explore it as either base or inspiration for possibilities on GODOT games running on rpi4 using the fb. I'll keep you posted if I get any results.

peicou, 2019-09-30:
Hi!
I have a rudimentary version of a non-X11 fb only port. It currently works with GLES2 only but next time I get some free time I'll add GLES3. It shouldn't be hard, just gotta figure out a better way to query the system as checking for bcm-host won't cut it anymore. No big deal.
The code needs a bit of clean-up but it works. I'm testing on a pi4 using raspbian lite, drm and gbm. 
If you want to check it out you can take a look at my repo:
https://github.com/peicou/frt_pi4
I'd love to add GLES3, polish it, test it, then add it to your project if it's something you'd like.
I have a test app that initializes a GLES3 context and loads the library, I just gotta make it play nice.
Otherwise let me know how to best credit you for FRT. It's really well done!

Thank you :)

efornara, 2019-09-30:
Thanks! It looks good to me. I'd love to include it in FRT.

To integrate it, I would need to ask you to change the name of the module. It's not really BCM / Pi4 specific, and ideally it should (be made to) work on other systems. For example, I plan to test it  on a PC with an Intel IGP. Something like `video_kmsdrm` probably makes more sense.

As for envprobe, I see if I can try something soon, hopefully already today/tomorrow. I don't know if testing the presence of the drm/gbm  libraries is the way to go. KMS/DRM really should be the default if you don't have X11. My idea is to parse one of `/proc/cpuinfo`, `/sys/firmware/devicetree/base/model`, etc...  on arm 32/64 and use the bcm module only in specific boards (there should only be a handful of them).

In any case, since probing is fragile by nature, I've been thinking of adding a `FRT_MODULES` environment variable to override the probing anyway, e.g.:

    export FRT_MODULES=video_fbdev,keyboard_linux_input,mouse_linux_input

peicou, 2019-09-30:
The environment variable would be the easiest to implement. I can work
FRT_MODULES into my branch. I can also focus on getting GLES3 up while you
get the system in place (replace the probing/access/etc?) and I'll adopt
whatever you push once the pi4 module is in a better state.

I agree with the naming as well. I created and named the module when I was
not sure about much and was just learning about the system. I'll be happy
to rename it to something that makes more sense.

Awesome! I'll be happy to contribute to FRT.

I'm ignorant about github/OpenSource etiquette. I cloned your repo then
pushed it to mine. The last thing I want to do is claim credit or anything
like that. Is there anything I should do so it's easier to contribute to
your project?

efornara, 2019-09-30:
I've pushed a new version of  `envprobe.cpp` (including handling of FRT_MODULES). I haven't tested it extensively yet (only vc4 with/without x11 on Pi3), but it looks OK.

Don't worry about attribution. If anything, the problem is the other way around: you should probably add your name(/email) after the FRT boilerplate in your main file (`video_kmsdrm.cpp`).  I see a minor issue with the mechanics of merging your patch, though. I don't see your repository as a fork of frt. That is probably for the better, since squashing several commits might be a bit tricky and, IIRC, would have involved a rebase.

_When it is time to integrate_, the best way is probably for you to make a fork of frt (IIRC you should be able to do it from the github web site) and clone it. You can then generate a diff  from _frt_pi4_

    .../frt_pi4 $ git diff 9ddc3a74958 >/tmp/diff.txt

(or whatever upstream commit you end up merging with), and apply it in _frt_.

    .../frt $ git apply /tmp/diff.txt

You can then open a PR with a single commit containing all your changes.

peicou, 2019-09-30:
that was fast! sounds good. I usually make time for working on my projects
over weekends but I'll try and get something done earlier. Thank you for
your advice, I'll use the new envprobe for the pi4 work and ping you when
it's ready.

Cheers!

peicou, 2019-10-02:
Hi!

I cleaned up some and started implementing GLES3. I got to a point where it
works if I pass --video-driver GLES3 but segfaults if I don't.
I can execute exported games both in GLES2 and GLES 3 mode so that's pretty
cool.

I Forked and pushed to https://github.com/peicou/frt if you want to check
it out, but I still want to figure out some things before I open an actual
PR. I'd love your suggestions for the following:

In both SCsub and detect.py, what would you use to figure out if the host
OS is a pi4? currently they look for bcm_host.h. Should I do something
similar to what you did in envprobe.cpp, or is there a better python-y way
to do it? I've never used python before and I thought I should ask before
trying something out.

Anyways it's looking good and I'm excited to contribute to your project!

thanks,

Paco

efornara, 2019-10-02:
I see that you went with the frt fork now. It is cleaner, but please keep in mind that I will accept a PR only if it is as a single commit, so you will have to squash all the commits before I can merge it.

The host OS shouldn't matter. You should be able to compile on a Pi Zero if you wanted. I detect `bcm_host.h` for `video_bcm.cpp` because that is the header that it needs to compile. You should detect whatever header your module needs.

efornara, 2019-10-02:
I did try your fork. Congratulations! It works on my PC and my Pi3! The only serious problem I had was the hard-coded `/dev/dri/card1`. Both on my PC and on my Pi3 I had to use `/dev/dri/card0`. I don't know if there is a correct way to find out which card to use. If you don't find a better solution, one idea could be to use `/dev/dri/card1` if it exists, and `/dev/dri/card0` if it doesn't. You can use something like this for the test:

    if (access("/dev/dri/card1", R_OK) == 0) { /* card1 exists */ }

In any case, it is probably better to add an environment variable that, if defined, overrides the choice made in the module, e.g.:

    export FRT_KMSDRM_DEVICE=/dev/dri/card0

I've spotted a few minor issues. Probably you were addressing them as part of the cleanup, but, anyway, here the changes that I ended up making:

```diff
diff --git a/detect.py b/detect.py
index d51cbc4..2f42712 100644
--- a/detect.py
+++ b/detect.py
@@ -197,7 +197,6 @@ def configure(env):
 	#if os.path.isfile('/opt/vc/include/bcm_host.h'):
 		#env.Append(FRT_MODULES=['video_bcm.cpp', 'dl/bcm.gen.cpp'])
 	env.Append(FRT_MODULES=['video_kmsdrm.cpp', 'dl/gbm.gen.cpp', 'dl/drm.gen.cpp'])
-	env.Append(LIBS=['drm', 'gbm'])
 	env.Append(FRT_MODULES=['dl/gles2.gen.cpp'])
 	if version.major >= 3:
 		env.Append(FRT_MODULES=['dl/gles3.gen.cpp'])
diff --git a/dl/egl.dl b/dl/egl.dl
index eaff378..573665e 100644
--- a/dl/egl.dl
+++ b/dl/egl.dl
@@ -24,7 +24,6 @@
  */
 
 #include <EGL/egl.h>
-#include <EGL/eglext.h>
 
 typedef EGLContext (*___eglCreateContext___)(EGLDisplay dpy, EGLConfig config, EGLContext share_context, const EGLint *attrib_list);
 typedef EGLBoolean (*___eglChooseConfig___)(EGLDisplay dpy, const EGLint *attrib_list, EGLConfig *configs, EGLint config_size, EGLint *num_config);
diff --git a/dl/gbm.dl b/dl/gbm.dl
index 86199ae..2f75188 100644
--- a/dl/gbm.dl
+++ b/dl/gbm.dl
@@ -23,6 +23,7 @@
  * SOFTWARE OR THE USE OR OTHER DEALINGS IN THE SOFTWARE.
  */
 
+#include <stddef.h>
 #include <gbm.h>
 
 typedef struct gbm_bo *(*___gbm_surface_lock_front_buffer___)(struct gbm_surface *surface);
diff --git a/video_kmsdrm.cpp b/video_kmsdrm.cpp
index aa0a0f2..35a4f6f 100644
--- a/video_kmsdrm.cpp
+++ b/video_kmsdrm.cpp
@@ -183,13 +183,13 @@ public:
         else
         {
             _resources = drmModeGetResources (_device);
-            if (_resources == nullptr)
+            if (_resources == 0)
             {
                 printf("failed to get resources\n");
                 return;
             }
             _connector = _find_connector (_resources);
-            if (_connector == nullptr)
+            if (_connector == 0)
             {
                 printf("failed to get connector. no fb?\n");
                 return;
@@ -284,7 +284,7 @@ private:
         if (!_initialized)
             return;
 
-        if (_crtc == nullptr)
+        if (_crtc == 0)
         return;
 
         drmModeSetCrtc (_egl.getDevice(), _crtc->crtc_id, _crtc->buffer_id, _crtc->x, _crtc->y, _egl.getConnector_id(), 1, &_crtc->mode);
@@ -355,7 +355,7 @@ public:
     int get_window_height() 
     { 
         int h = 0;
-        if (_crtc != nullptr)
+        if (_crtc != 0)
         {
             h = (int)_crtc->height;  
         }     
@@ -365,7 +365,7 @@ public:
     int get_window_width() 
     { 
         int w = 0;
-        if (_crtc != nullptr)
+        if (_crtc != 0)
         {
             w = (int)_crtc->width;  
         }     

```

I actually had to re-add `-lgbm` otherwise it wouldn't link, but it shouldn't be there.

I'm afraid you cannot use nullprt: frt is C++98/03. I don't like the 0s (I wanted to test quickly, and it was easier not to make mistakes that way). NULLs would have been even worse. The best way to test pointer is probably `if (ptr)` or `if (!ptr)`.

Sorry to nitpick, but one thing that I've noticed is that you've changed class members to begin with an underscore. I'd appreciate if you revert that. It might not be strictly against the C++ standard, but it is close enough to be problematic. It is probably better / easier to just leave names beginning with an underscore to the language/compiler/OS. https://stackoverflow.com/questions/228783/what-are-the-rules-about-using-an-underscore-in-a-c-identifier

peicou, 2019-10-03:
Thank you for your feedback! It's been a few busy days but I'll keep it all
in mind as soon as I dig back in.
I'll squash everything into a single commit PR.

peicou, 2019-10-04:
I made some time today and implemented/pushed most of the changes you
suggested.
I have a few questions-
1) I removed drm, but found the same issue when removing gbm. For the time
being I left it there but will gate it so it's only added when using
kmsdrm. Is that a viable solution?
2) What's the issue with including eglext.h? I need it to create attribute
array for ES3 (EGL_OPENGL_ES3_BIT_KHR).

Other than that, I have a short to-do list before I open a PR-
>Make sure it works OK with both ES2 and ES3
>Figure out the gating to append kmsdrm CFLAGS so it plays nice with the
rest of the system (specifically so it doesn't clash with bcm)
>Figure out why I'm getting the following error:
Main: Load Modules, Physics, Drivers, Scripts
ERROR: PhysicsServer: Condition ' singleton != __null ' is true.
   At: servers/physics_server.cpp:726.
Main: Load Translations
A quick google sent me to some of your posts before. Did you ever figure it
out?
I'm open to other suggestions if you have any.

Thank you for your time and effort maintaining FRT!

efornara, 2019-10-04:
**gbm**

I'm sorry, but linking it at compile time is not an option. I'm not sure if lite comes with it by default, but even if it did, FRT might be used in other (old) distros. On hardware other than the Pi, some legacy driver might only work with an old linux kernel / distro. You went all the trouble to define the symbols for drm / gbm already. It would be a shame to have wasted all that effort! I didn't check which ones are missing, but isn't likely to be just 1 or 2 at this point?

**eglext.h**

If you *really* need it, use it, but please read on first.

When I checked the diff of changes, I saw eglext.h and I thought: "That might be a problem. If he needs an extension, how does he check if it is supported, and what does he do if it isn't?".  I first tried to comment it out and it compiled and run without problems, so I thought maybe it was an oversight. It turns out that I compiled with Godot 2.1 and I didn't notice that you were using it to ES 3.

Did you have to add EGL_RENDERABLE_TYPE, or has it always been there? Have you tried to leave it out? Both on my PC and on my Pi 3, eglinfo doesn't show configs that are specific for ES3. If we can get away without using extensions it would be better. In any case, we can change the visibility of EGLBaseContext::build_attr_list to protected and you can use that, so you would get the params that you are currently missing (e.g. multisample).

**Appending kmsdrm CFLAGS**

I haven't tested it, but this should do:

    if os.path.isfile('/usr/include/gbm.h'):
        env.Append(FRT_MODULES=['video_kmsdrm.cpp', 'dl/gbm.gen.cpp', 'dl/drm.gen.cpp'])

Leave the bcm module alone (i.e. don't comment it out). They are independent, so, when it comes to compiling them, any combination is fine: bcm, kmsdrm, bcm+kmsdrm, none. Which one to probe is decided at run-time. I even compile in the bcm module on my PC! (of course there is not way it would run, but it is never probed anyway).

**ES2 / ES3**

I can try to give you some hints.

Check video_x11.cpp. It looks like you have taken some bits of code from video_x11.cpp already. If you re-read it now, after you have some more experience with the codebase, maybe it makes more sense. Take the include block and the frt_load_gles definition as *they are*, including the weird indentation. FRT_GLES_VERSION is not used for ES driver selection, but, in a way,  for *godot version* selection. In godot 2.1 (and maybe godot 4.0/5.0 if they decide to scrap GLES 3 altogether), you cannot include/compile GLES3 , because it is not available.

In the initialization code (EGLKmsdrmContext::init), you have to use the version that is passed to you by the engine. You cannot use FRT_GLES_VERSION. If you re-read frt_load_gles in video_x11.cpp now, you should notice that it too is actually using the version passed in; it is not using it only if version == 3 is not an option anyway (e.g. godot 2.1/5.0, when it is always == 2).

**PhysicsServer**

To be honest I've never checked. My original post upstream was about rendering performance, and I took https://github.com/efornara/frt/issues/9 as being about compiling templates. Anyway, I'll have a look. I wouldn't hold the merge for that, though. It isn't related.

peicou, 2019-10-06:
Hi,

master in my fork implemented all of your suggestions.
I also checked the PhysicsServer error- Godot is initializing the 3d
physics server twice. I decided to stop digging as the error message is a
singleton "heads up" type of protection so it shouldn't matter too much.

I think it's ready for a PR, let me know what you think.

Thank you,

efornara, 2019-10-06:
gbm: perfect

eglext.h: I see you've kept a line EGL_RENDERABLE_TYPE, EGL_OPENGL_ES2_BIT.  That is passed even on ES3. Isn't it better to remove it? Have you tried removing it?

CFLAGS: looks good

ES2 / ES3: You are still not using the "int version" passed in "EGLKmsdrmContext::init" (run-time selection), but you rely instead on FRT_GLES_VERSION (compile-time selection).

Another thing that I've noticed reading it more carefully is printfs. Some are not needed, some should be "fatal".

Let me know if want to have another go, or you prefer to merge it as it is and let me handle it.

peicou, 2019-10-08:
I'll get it to a better place! Thank you for your patience and feedback.

egltext.h- I'll remove EGL_RENDERABLE_TYPE
ES2/3- I'll fix that.
printf/fatal- I'll address this too.

Work is getting loaded but I'm pushing those changes at the end of the week
at the very latest. I'm hoping I can do it way sooner, I do want to change
and test before pushing.

Thank you!

efornara, 2019-10-09:
I'm glad that you've decided to stick to the end.

Don't worry about finishing as soon as possible. One or two weeks doesn't make much of a difference.

Please allow for some time for your PR to go through. I am not so sure it will go smoothly at the first try. I have seen that you've "rewritten" some of my changes in your master branch instead of merging them from upstream using git. I assume that you haven't added an "upstream" remote to your local git repository yet.

While this is written for the engine itself, I find it is a good guide overall:

https://docs.godotengine.org/en/3.1/community/contributing/pr_workflow.html

Finally, while I don't mind that you've chosen an indentation style that is different from the rest of frt, when making a diff, I have noticed that there are some trailing whitespaces. They might cause problems in the future. You should be able to see them by typing something like:

    git diff b9d54a1 --check

peicou, 2019-10-11:
Thank you for your patience, I appreciate your help as I get familiarized
with the process.
I (tested and) pushed more suggested changes to my frt fork under kmsdrm.
Hopefully during the weekend I'll open up a PR following the procedure
suggested here:
https://docs.godotengine.org/en/3.1/community/contributing/pr_workflow.html

Have a great weekend!

efornara, 2019-10-13:
Thank you also for your patience. I have looked at your branch right now. Code-wise, It should be merge-able after these changes:

1) Overwrite release.sh and envprobe.cpp with the ones taken from my branch.

2) Remove indentation changes of existing code in SCsub. The easiest way is probably to overwrite the file with the one taken from my branch and then reapply your change:

        if os.path.isfile('/usr/include/libdrm/drm.h'):
            frt_env.Append(CCFLAGS=["-I/usr/include/libdrm"])

3) Add missing newline at the end of file dl/gbm.dl

4)  Remove fatals from the probe method. This method, unlike init, should be silent and simply return false if the module cannot be initialized.

Procedure-wise, I don't know. I hope that git and github are smart enough not to complain about code that ended up looking the same even if it comes from a different history. In any case, the important thing is that the code is fine. It the worst case, you can always make a diff and reapply it a against an updated branch from upstream.

peicou, 2019-10-13:
Thank you for the feedback!
changes applied and merged to my master! Would you say it's ready for a PR
now? I'm happy to fine tune it some more. I also tested w GLES2 and GLES3
and it works fine with a basic 2D Project on raspbian lite for pi4.

efornara, 2019-10-13:
Almost.

Only fatals remain a problem. Maybe I wasn't clear. This is what I meant:

- `bool probe()` should be silently returning false on errors. This is OK in your branch now.
- `void init(int version)` should be calling fatal, not simply return. This was fine before, but now it isn't.

Everything else looks fine.

peicou, 2019-10-15:
Hi,

could you please take a look at my frt fork? let me know if there are any
other changes required.

Thank you

efornara, 2019-10-15:
Thanks! It looks good.

There might be a couple of spots left, but I think it is ready for a PR.

Again. while code-wise it is fine, github says "This branch is 20 commits ahead of efornara:master. ". Ideally, a PR should have only a single commit.

To give you some ideas, here is how I think I would do it:

    cd ~
    git clone https://github.com/peicou/frt test_pr
    cd test_pr
    git remote add upstream https://github.com/efornara/frt
    git fetch --all
    git diff upstream/master >~/changes.patch
    git checkout -b kmsdrm_pr upstream/master
    git apply ~/changes.patch

You should then be able to add the files, commit and push.

This newly created branch (kmsdrm_pr) should be OK to merge with upstream.

peicou, 2019-10-17:
done! single commit.

Let me know!

Paco

efornara, 2019-10-17:
Merged. Thanks!

I haven't tested this commit yet, but I've tested a previous version and it was working, so I don't expect any surprises there.

closed 2019-10-17 by efornara