efornara/frt #13: Cannot save or load from user:// with FRT exports on Raspberry Pi 4

fornclake, 2019-10-01:
Saving and loading files from the user directory does not seem to happen at all with FRT. Game does not crash, it just does not seem to write any files anywhere. Are FRT-exported games able to write to disk?

Running the game through an X11 export seems to save and load but I can't get it to work above 5 fps on my Raspberry Pi 4. FRT has it running perfectly, just without the saving or loading.

efornara, 2019-10-02:
Something got lost in the Godot 2 to 3 transition. Since it doesn't cause compilation errors, it went undetected. I should be able to fix the source soon, but I am not planning to generating new FRT binaries until the new version of FRT is released (I guess in a couple of weeks).

It has been a while since I have tested the X11 platform (Godot 3.1), but I don't remember such a drastic drop in performance (Pi 3). Have you followed my tips on how to compile it? https://github.com/efornara/frt/blob/master/doc/Compile.md There might be a few reasons to use FRT on a Pi4 (e.g., AFAIK, GLES3 doesn't work on the X11 platform), but performance shouldn't be one of them.

fornclake, 2019-10-03:
Glad to hear you have a fix for FRT. Even if the Pi 4 doesn't necessarily need it, I would like to try exporting Godot games to the 2 or 3 to test performance.

As for the X11 platform, nothing I've tried seems to be working. I tried compiling using the settings on [this page](https://bits.krzysztofjankowski.com/raspberry-pi-4-as-perfect-indie-console/) but it runs at an extremely low fps. When I tried using your settings for Pi 2/3 on the Compile page, the game crashes soon after starting. I am compiling on the Pi itself, and the game uses GLES2. Do you have any other things I could try? Otherwise FRT seems like it will do the job.

efornara, 2019-10-03:
The fact that my compile instructions cause you problem is concerning: it might be an accident that FRT works at all, and any new release might stop working. By crash you mean segmentation fault? Do you remember the command line that you used, or at least if you used gcc or clang?

As for the X11 platform, the easiest thing you can try, if you didn't already, is to change some of the settings in the page that you linked:

    target=release
    tools=no

The settings as shown in that page are supposed to generate somewhat slow binaries.

Other options (llvm with thinLTO or, the one that I use, jessie chroot and gcc 4-9) are a bit more involved.

Anyway, I don't want to sound like I am not recommending FRT. If it works for you, of course I am happy for you to use it.

While I am not going to build all the flavours now, I can build and upload a single one if you need it. Just let me know which one. Godot 3.1.1 for pi2?

fornclake, 2019-10-03:
I'm not sure what the crash was exactly. The title screen for my game loaded with your compilation instructions, but loading into a new screen crashed. I used clang. I didn't have all of the required libraries so I used the one-liner for Debian [here](https://docs.godotengine.org/en/3.1/development/compiling/compiling_for_x11.html).

I have also tried compiling from that blog post with target=release and tools=no the other day. It made it past the title screen but still had the 2-3 fps.

If you can build Godot 3.1.1 for pi2 that would be super great! I was going to try to work around the lack of save&load if I couldn't figure the X11 export out, but having a working FRT binary can help me out a ton. Thanks so much for your help (and FRT)!

efornara, 2019-10-03:
I've just uploaded the binary for the latest frt commit to sourceforge (demos). I haven't tested it.

fornclake, 2019-10-06:
Sorry it took me a second to get back to you, the new binary is working great and save and load works. Thank you so much!

Just one last question for now, have you had any issues with crackling/garbled sound on the Pi before? The only other person I've seen with that issue on other people's Godot/Pi projects was a Raspberry Pi 4 project with FRT as well (not shown, but briefly mentioned in [this video](https://www.youtube.com/watch?v=yqYhB4udVGU)). If it's just a Pi 4 thing there might be some digging I can do to see if it's a sound card driver issue, and I think I'm going to use my Pi 4 for something else and grab a 2 or 3 for Godot anyway. If it's something you've seen before, do you have any suggestions or is that just a limitation of Godot on Pi?

efornara, 2019-10-06:
My Pi is hooked to a monitor (no-sound). I almost never test in on a TV (with sound).

The last time that I tested sound, the quality wasn't that good (on a Pi3, I don't have a Pi4 yet), but I don't remember being that bad either.

I don't think there is much I can do: I instantiate the ALSA linux godot upstream driver as it is. I'll probably re-test the sound and re-check the source one of the next few days. If/when you happen receive a 2 or 3, I would appreciate if you could comment on the new issue that I have made (https://github.com/efornara/frt/issues/14) and let me know if the problem is limited to the Pi 4.

closed 2019-10-17 by efornara