efornara/frt #14: Poor Sound Quality (on Pi4?)

efornara, 2019-10-06:
There might be an issue with crackling/garbled sound on the Raspberry Pi 4, and possibly other Pis. More testing / feedback is needed to see which versions (of Pi and Godot) are affected.

See https://github.com/efornara/frt/issues/13 and also https://www.youtube.com/watch?v=yqYhB4udVGU .

hiulit, 2020-11-17:
I have a Raspberry Pi 4 and I'm also getting crackling/garbled sound using FRT 0.98 (Godot version 3.1.2). I've also compiled my own export template using this guide https://bits.p1x.in/raspberry-pi-4-as-perfect-indie-console/ and I also get poor sound quality.

I compared the same game https://github.com/securas/SealedBite and when I run it on my Mac, the sound is perfect, but when I run it on the Raspberry Pi 4, it's awful.

charasyn, 2020-11-20:
Hello,
I am working on a game which is targeted at running on Raspberry Pi 3.
Using a brand new install of Raspbian 32-bit version, I tried to run my game exported from Godot 3.2.3 using "frt-098-323-pi2.bin". Thankfully, the graphical performance is excellent and runs at a fluid 60fps! However, I am having an issue with sound using both the HDMI and audio jack.

I see constant errors in the console when running the game: `ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred`
This seems to imply that an audio buffer underrun is occurring, which I think means that the Godot game is not supplying audio as quickly as ALSA is consuming it.

Some more useful information:
By checking `/proc/asound/Headphones/pcm0p/sub0/status` while the game is running, I can see the following information:
```
access: MMAP_INTERLEAVED
format: S16_LE
subformat: STD
channels: 2
rate: 8000 (8000/1)
period_size: 256
buffer_size: 258
```

2 takeaways here:
1. The audio rate is only 8KHz! That would sound bad even without buffer issues.
2. The buffer_size is only slightly bigger than the period_size, which I think means that it is very likely that this type of buffer underrun should occur...

For reference, here is similar output while playing back audio with VLC:
```
access: RW_INTERLEAVED
format: S16_LE
subformat: STD
channels: 2
rate: 48000 (48000/1)
period_size: 1920
buffer_size: 32768
```

We can see that this uses a much larger buffer than one period of audio. Audio playback with VLC is clear and crisp as expected.

As far as I can tell from the code, FRT doesn't do anything special with ALSA; it seems that the main Godot code is used for ALSA, so maybe there is a bug in there that's causing it to use invalid settings? (Or maybe the audio interface isn't being initialized properly from FRT? I haven't looked that deep so I have no idea)

Anyways, it seems like this issue might affect more than just RPi4. I hope that this information helps debug this issue.

charasyn, 2020-11-20:
New information!

By changing my project audio mix rate from "44100" to "48000", I was able to get the sample rate on the Pi to go from 8000 to 48000. Additionally, a combination of doing this and changing my audio latency from 15 to 20ms caused the period_size and buffer_size to go from (256, 258) to (1024, 2048). This fixed the issue with the audio buffer underrun.

However, unfortunately it still sounds bad. I think Godot might be providing the samples in the wrong order or something, because audio that should only be heard through the left ear is actually heard through both. I don't have time right now to delve further into Godot's ALSA code unfortunately, but looking at the main driver [here](https://github.com/godotengine/godot/blob/master/drivers/alsa/audio_driver_alsa.cpp) would be my next step.

I also wonder if ALSA is broken on desktop? This could be causing the issues maybe. I unfortunately also can't check that right now.

hiulit, 2020-11-20:
Great information @charasyn ! Maybe we should take this to Godot's issues?

I've done some more quick tests. I've installed Godot on an x86 Debian virtual machine and the sound is perfect. So it might be indeed something related to the Pi's only.

charasyn, 2020-11-27:
Hi,
I think I have diagnosed this issue. It appears to be a bug in Godot's ALSA driver. Relevant code:
```cpp
while (todo && !ad->exit_thread) {
    uint8_t *src = (uint8_t *)ad->samples_out.ptr();
    int wrote = snd_pcm_writei(ad->pcm_handle, (void *)(src + (total * ad->channels)), todo);

    if (wrote > 0) {
        total += wrote;
        todo -= wrote;
    } else if (wrote == -EAGAIN) {
        ad->stop_counting_ticks();
        ad->unlock();

        OS::get_singleton()->delay_usec(1000);

        ad->lock();
        ad->start_counting_ticks();
    } else {
        wrote = snd_pcm_recover(ad->pcm_handle, wrote, 0);
        if (wrote < 0) {
            ERR_PRINTS("ALSA: Failed and can't recover: " + String(snd_strerror(wrote)));
            ad->active = false;
            ad->exit_thread = true;
        }
    }
}
```
It appears to only be triggered when the ALSA function `snd_pcm_writei` is unable to accept the entire sound buffer at one time. When this happens, the `wrote` variable is updated with the number of samples accepted into the buffer, which then updates `total`. `total` is used to adjust the index into the sample buffer so that we don't try to send the old data to the sound device, but this is where the bug lies.

Since *src is a uint8_t pointer in the code, when doing `src + (total * ad->channels)` this adjusts by that amount of bytes. However, each sample is a 16-bit number, meaning that the source pointer is adjusted by the incorrect number of samples. This can also result in audio data intended for the Left channel being sent to the Right and vice versa, which is why I thought it might be something to do with indexing into the buffer.

I'm going to make a pull request in the main Godot repository to fix this, but for now, I would recommend changing the type of *src from `uint8_t` to `int16_t` to match up with the actual data. This fixes the issue on my end.

hiulit, 2020-11-28:
I've compiled Godot 3.2.3 on a Raspberry Pi 4 with that changed and it fixed the issue! :)

hiulit, 2020-11-30:
@efornara will you wait until an offical release from the Godot team to recompile your binaries with the audio fix? I've been able to compile 3.1.2 and 3.2.3 with the fix and they work perfectly.

efornara, 2020-11-30:
@charasyn Thanks! I haven't tried it yet, but it looks good.

@hiulit I am importing the patch into the FRT repository, and applying it myself to every versions of Godot that FRT supports. The idea is to release a new version of FRT; ideally today, but I am not 100% sure.

efornara, 2021-01-11:
I have finally tested the patch with an headphone and I can confirm that it works for me too.

I don't actually have problems with 44100Hz (Pi 3B+ without pulseaudio installed). Checking in /proc, it seems that both 44100Hz and 48000Hz are fine. Default latency (15ms) is too low also in my tests.

I've uploaded a couple of official 3.2 demos on sourceforge (2d platformer and 3d dodge the creeps) and just to be sure I've changed the audio settings to 48000Hz and 30ms.

closed 2021-01-11 by efornara