efornara/frt #17: FRT forcing "display/window/stretch/mode" to "viewport"?

hiulit, 2020-02-06:
I have a project that uses the 2d stretch mode, but when I open it on a Raspberry Pi using the latest version (0.9.7) of FRT, it seems like it is forcing the stretch mode to viewport.

He's a comparison of the project I'm working on.

These images are just an example to see the differences between stretch modes. When I open the project with FRT, I see the second image. That's why I think FRT might be doing something to the stretch mode.

## display/window/stretch/mode -> 2d
[image]

## display/window/stretch/mode -> viewport
[image]

efornara, 2020-02-06:
FRT shouldn't force a particular stretch mode.

The problem is that FRT always opens a window of exactly the requested size, so stretch mode is in fact (almost) never used. That sort of looks like viewport mode, even though it is technically very different.

I do plan to add a custom param that makes FRT follow more closely the way Godot/X11 works, and, after a transitional version, for it to become the default. Initially it was meant to happen starting from 0.9.7, but I am now planning to do it a bit later.

The way FRT currently works has some performance benefits, most notably in 3D, that is why FRT started that way. For example, I can get a solid 60 fps at 960x540 on a fairly complex 3D scene on a Pi 0/1. I can only get 30 fps by using the Godot/X11 way, i.e. rendering at 1920x1080 or rendering at  960x540 and upscaling it to 1920x1080 using OpenGL (stretch mode set to viewport).

All the stretch modes should still be available even in the current versions of FRT, though. For the time being, you should be able to access them by setting test_width and test_height.

For example, assuming a 2D pixel game designed at 200x300:

    size: 200x300
    test size: 400x600
    stretch mode: 2D

Changing the aspect ratio should also work:

    size: 200x300
    test size: 400x800
    stretch mode: 2D
    stretch aspect: keep width

Let me know if this doesn't work as a temporary fix for you.

hiulit, 2020-02-07:
Hey, thanks for that extensive reply! But I don't know I follow you... sorry :)

Here are my project properties:
```
size: 224x288.
test size: 448x576 (2x)
stretch mode: 2d
stretch aspect: keep
```
And this doesn't look good on a 1920x1080 screen.

What you are suggesting is that if I use a 1920x1080 screen I should change the project's test size to try to match it to the screen's size? Something like 672x864 (3x) or 784x1008 (3.5x)?

Thanks!

efornara, 2020-02-07:
> What you are suggesting is that if I use a 1920x1080 screen I should change the project's test size to try to match it to the screen's size? Something like 672x864 (3x) or 784x1008 (3.5x)?

Pretty much.

With this:

    size: 224x288.
    test size: 448x576 (2x)

FRT (bcm) renders at 448x576 and let the hardware scaler upscale it to 840x1080 (using linear filtering).

If you want something different, you have to take charge of the upscaling and weight the trade-offs yourself. I expect that, even after the planned change to FRT to follow X11 more closely, you would still have to take charge of the trade-offs.

I mean, what happens with standard Godot/X11? It looks to me that you would get a 448x576 window that you can then manually resize to a size that you like (e.g. 672x864, 784x1008 or something else). Even fullscreen doesn't really change that.

In your case, I would probably try with something like this:

    size: 224x288.
    test size: 672x1080 (pixel perfect - 3x + top/bottom black border)
    stretch mode: 2d
    stretch aspect: keep height

or this:

    size: 224x288.
    test size: 840x1080 (fullscreen)
    stretch mode: 2d
    stretch aspect: keep

and see how they look.

That of course assumes that this is not a bug in FRT to begin with, but unless I remember it wrong, FRT doesn't touch the stretch options at all.

hiulit, 2020-02-12:
I used

```
size: 224x288.
test size: 840x1080 (fullscreen)
stretch mode: 2d
stretch aspect: keep
```

and it seems to be working :) Thanks!

It still doesn't look the same on a PC (or Mac) as on a Raspberry Pi 3 though, but the improvement is substantial! They look practically the same. Maybe it has to do with the shader I'm using.

When I use Godot/X11 using my initial settings

```
size: 224x288.
test size: 448x576 (2x)
stretch mode: 2d
stretch aspect: keep
```

I can resize the window and everything looks good at any resolution. But when using that same settings with FRT on a Raspberry Pi, it looks like I showed you in the previous post.

Well, now I know that I have to set the test size the desired output resolution.

efornara, 2022-01-29:
Closing all FRT 1 issues. Please comment on https://github.com/efornara/frt/issues/39 if you get a chance to try again with the latest 2.0 binaries, and the problem is still there.

closed 2022-01-29 by efornara