efornara/frt #18: segmentation fault error

devops724, 2020-02-28:
Hi there
i have issue on convert game for arm
i export project into pck/zip (unchecked x64)
move project into raspberry and run 
frt_097_320_pi2.bin --main-pack doge.pck
return this 
Godot Engine v3.2.stable.custom_build.4e7d75ccd - https://godotengine.org
Segmentation fault
what is wrong ?
i also try -d argument for debugging , same output
rpi3 
debian os

efornara, 2020-02-28:
> Segmentation fault
> what is wrong ?

You cannot really tell; there are too many possible causes for a generic error like "Segmentation fault".

You could eventually rebuild a debug version of the binary and try to investigate the problem, but first it makes more sense to make sure that the problem is FRT-specific. If it isn't, the upstream repo has many more eyeballs to help you track down the problem.

So, I would suggest:

1. Test with the experimental OpenGL driver (vc4), i.e. x11 instead of bcm
2. Test with the standard Godot runtime too, i.e. official upstream source compiled with platform=x11

Once you do that, we'll know if it is better to continue here, or open an issue upstream.

Let me know if you have any problems compiling a standard x11 runtime yourself.

daviel, 2020-03-03:
I had the issue with the same error code on my raspberry zero and godot 3.2. You have to check etc in the export template. After this it worked for me.

devops724, 2020-03-04:
> I had the issue with the same error code on my raspberry zero and godot 3.2. You have to check etc in the export template. And don't forget to enable GLES2 in the project. After this it worked for me.

it worked for me, now it run on rpi3 , 
now keyboard not working , actually no keyboard and mouse events triggered on godot game
is there any other magic checkbox to solve this?

daviel, 2020-03-04:
> it worked for me, now it run on rpi3 ,
> now keyboard not working , actually no keyboard and mouse events triggered on godot game
> is there any other magic checkbox to solve this?

Keyboard inputs were working out-of-the-box for me. You may have to check the godot project itself whether it maps the inputs correctly.

devops724, 2020-03-04:
can you test compile use this template 
Dodge the Creeps
it return segment fault

daviel, 2020-03-05:
> can you test compile use this template
> Dodge the Creeps
> it return segment fault

It worked for me immediately. With mouse and keyboard inputs and sound output. I have imported the project to godot 3.2 set it to GLES2 and ETC enabled. I also turned off every graphic intensive option in the settings. See screenshot.

[image]

closed 2020-06-20 by efornara