efornara/frt #19: automatic cross compile

daviel, 2020-04-07:
Hi,

I really like your project but thought there should be a process to improve the compilation of it with godot so I wrote a small bash script to automate cross compilation using qemu+chroot here: https://github.com/daviel/frt_builder

I have tested the cross compilation successfully on a pi-zero. It should work on every debian based distribution. I also included your binarys from soureforge to download from github in the release tab for more convience.

Let me know what you think and whether it would be a good idea to link to the project from frt or even merge this project inside frt.

Thank you!

BTW: There is currently a branch of godot for batching of 2D-rendering which increases the performance by a large margin. I have used the branch and tested it on the raspberry pi zero with FRT and could increase the FPS by about 1.8 times (30 -> 55 FPS)! See here for details: https://github.com/godotengine/godot/tree/3.2-gles2-batching

efornara, 2020-04-07:
Hi,

Your script looks a very good option if you want to compile godot/frt from scratch from a PC, and I encourage you to continue to improve it and to maintain it, because I suspect other people might find it useful. 

As for merging, to be honest, my needs sound different.

If upstream releases a new version, I consume very little bandwidth, don't spend too much time leaving my Pi3 to generate the new binaries, test only one of them, and release them all. In other words, I am very, very happy with my current process as it is.

I do plan to (try to) change it. In FRT current roadmap, that would be 1.0.x (buster / cross compile / flags / docker). But I do expect to spend several hours when I finally decide to do so. Just retesting as many combinations as I can (several boards and distros) sounds daunting. That is why I prefer it do it only once, and maybe keep supporting the old 0.9.x builds for a while. There is even a blocker now. I don't run RetroPie, but I consider it a very import distro, and, last time I checked, it stills uses debian scratch (i.e. older libc), so I will have to wait at least for them to upgrade to buster.

For what I can see, your script would not be optimal for my setup. For example, for me qemu is slower than native compilation (I use a very old chromebook as a building machine), and that is why I will try "native" cross compilation for FRT 1.0.x (at least from 2/3 of the build). Also, my connection is plain old ADSL, so downloading too much is a pain for me.

Where I think your builds can provide a lot of value to users is your support for batching. I am not interested in tracking an unmerged tree and I am happy to wait for the patch to be included in an official release. So, if you are interested, let me know if it is OK for me to link to your project on the FRT's README page to give it a bit of visibility.

daviel, 2020-04-08:
Thank you for your fast response.

I think this would be a good way to go. Linking to my project would be sufficient for most I think. I will keep the project up-to-date as I need it on my own currently so everyone who might want fast and off-branch builds can use it.

The cross-compilation takes quiet some time that's right. It took about 70 minutes to compile for a pi0 on my i7 which is really slow considering that a native x86 build does only take a few minutes. But it is much easier to setup and still faster than a build on a raspberry so I will stick with it. 

I did not keep in mind the limitations like a slow connection/machine so my script might only be useful for high-end machines.

Could you successfully activate LTO for your build? Currently I get an out-of-memory error in the linker when LTO is activated which is weird as I have more than enough RAM available.

efornara, 2020-04-08:
I can (and do) use thinLTO (for godot2-rpi-repo), but only for  2.1.x. I tried ages ago for 3.x and I also got an out-of-memory error in the linker.

Total memory available on the host can be deceiving. If you use qemu-static, you are using a 32-bit linker, so you are most likely limited by the size of its address space.

daviel, 2020-04-08:
That were my thoughts too. I will take a look at this as I would really like to know whether we can get any gains through LTO on the PIs.
Thank you so far. :)

closed 2020-04-08 by daviel