efornara/frt #20: Google Coral AI Dev Board

rac0316, 2020-04-09:
Hey, any thoughts on how FRT can run on the new Coral AI Board from google? Specs here:

https://coral.ai/docs/dev-board/datasheet/

Supports Open GL ES 2 and 3 - followed instructions to create a pck with no extra graphics options/different combinations, but when I run through FRT I get a segmentation fault.

efornara, 2020-04-09:
On paper, it looks like FRT should work there. Unfortunately, a segmentation fault makes it hard to figure out what is going wrong.

If you want to have a try, I would suggest a couple of tests as a start:

1. Install and test es2gear / es2_info (they should be in mesa-utils-extra) to see if they work.
2. Use a pack known to work from https://sourceforge.net/projects/frt/files/demos/ , for example:

    frt_097_216_pi1.bin -main_pack 21_2d_platformer.zip

rac0316, 2020-04-09:
I'll play around with this, thanks!

Tried a few different commands/examples and here are the results:

es2gears: Segmentation Fault
es2_info: Segmentation Fault
glxgears: Runs fine (not sure the difference between this and es2gears)
pygame2 test running glcube example: Runs fine  
Running your example: Segmentation Fault

Maybe there's just a missing driver somewhere?

efornara, 2020-04-09:
It looks like GLX / OpenGL is working and EGL / OpenGL ES is not.

If glxgears is fast enough / glxinfo tells you that OpenGL is accelerated, you can compile the official godot export template (i.e. platform x11, no need to use FRT at all) yourself and try that.

If OpenGL is not accelerated, you have driver issues and you should try to fix them first. If you get es2gears to run, you can then try godot/frt again. If you get glxgears to run *accelerated*, you can then try godot/x11 again.

rac0316, 2020-04-09:
It's saying accelerated (direct rendering: Yes), so I'll follow try the custom export template approach first! So the processor is ARM - is there still a way to do compile it myself? Any examples floating around?

Thanks for the help!

efornara, 2020-04-09:
It should say "Accelerated: yes". Another thing you can check is if the renderer is something like llvmpipe (i.e. non accelerated).

As for compiling the template yourself, the easiest way is probably to do it directly on the board:

https://github.com/efornara/frt/blob/master/doc/Compile.md (X11)

https://docs.godotengine.org/en/3.1/development/compiling/compiling_for_x11.html (export template)

rac0316, 2020-04-09:
Very cool - works nicely! Thanks for point me in the right direction..

For reference, I compiled Godot with the following for this board:
`scons -j 6 platform=x11 tools=no target=release_debug bits=64 use_llvm=yes use_lld=yes`

I then was able to take the built binary and use it as a custom export template for a Godot project from another computer!

closed 2020-04-09 by rac0316