efornara/frt #22: Calling OS.get_name() returns "Unix" rather than "FRT"

dsnopek, 2020-07-07:
I decided to make a Raspberry Pi version of my game, and wanted to disable some things (ex. particles) when running on the Pi, and so I figured I'd use `OS.get_name()` in GDScript to detect when running under FRT. Looking at frt/os_frt.cpp I would have expected it to return "FRT":

https://github.com/efornara/frt/blob/master/os_frt.cpp#L562

However, instead, it returns "Unix" which I guess I can still use, but it feels imprecise (ie. there isn't another "Unix" platform that I'm targeting - Linux will return "X11" here - but it'd still be nice to get "FRT")

I'm using Godot 3.2.2 with FRT 0.9.7 (the pre-built binaries from Sourceforge in frt_097_322.zip)

efornara, 2020-07-13:
Thank you. It should be fixed now. I am not tagging a new FRT release, but you can find the new binaries in https://sourceforge.net/projects/frt/files/previews/ (named after commit https://github.com/efornara/frt/commit/d5e8418e67d285305be47bfa0028efd5af2c868f).

dsnopek, 2020-07-13:
Thanks so much!

closed 2020-09-18 by efornara