efornara/frt #23: Store the precompiled FRT binaries on GitHub instead of on SourceForge

hiulit, 2020-08-07:
When you create a new release for FRT, you can add the precompiled binaries as "assets":

[image]

So the binaries get attached to a specific release:

[image]

*This is an example of what I'm using in my project https://github.com/hiulit/RetroPie-Godot-Game-Engine-Emulator/releases*

It's easier for the users to find them, and I think it's easier to maintain, for you :)

efornara, 2020-09-18:
I don't find binaries in github so much better than binaries in sourceforge.

The main problem I have with sourceforge is a few distracting ads, but when it comes to discoverability, I am happy with it:  there is a big green button linking directly to the "don't make me think" archive. Also, there is no obvious place where to put demos and previews in github. Finally, once in a while, I am curious to check the stats. Not to mention that all the existing binaries archives have already been uploaded there.

This issue has been open for more than a month now with no comments or upvotes; I take it as a hint that most people can live with the way it is now.

Sorry, but, for the time being at least, I prefer to keep releasing on sourceforge.

hiulit, 2020-11-17:
It's fine. No worries. Whatever works best for you.

closed 2021-01-11 by efornara