efornara/frt #25: What would the first steps be for porting FRT/Godot to an older MIPS chip? (JZ4770 specifically)

rcoppy, 2020-11-11:
Hi there! This looks like a really fantastic project; amazing what you've got going. 

Reading through the docs and discussions it looks like a lot of time has been taken to figure out a solution for building to Raspberry Pi (ARM). Another popular device with the single-board computer gaming community is the RG350; the newer RG351 runs on an RK3326 chip (Quad-core ARM Cortex-A35 CPU), but the older, more widespread model is built on the JZ4770, which is a MIPS chip. 

The RG350 and its other MIPS siblings are all based on the open-source but defunct GCW0, whose stock firmware is a Linux variant called OpenDingux. 

There exists a somewhat dated [toolchain for developing OpenDingux.](http://www.gcw-zero.com/develop)

The JZ4770 SOC comes with a Vivante GC860 GPU, which has an open-source driver available [named Etnaviv](https://github.com/etnaviv/etna_viv) that appears to support DRM/KSM.

User @ppiecuch [did a preliminary build](https://www.reddit.com/r/godot/comments/igw7iz/building_godotengine_with_gcw0_toolchain/) of Godot/FRT using the GCW0 toolchain, which unfortunately doesn't seem to yet work. 

Here are their relevant forks: 
https://github.com/ppiecuch/frt
https://github.com/ppiecuch/godot

I unfortunately am not all that familiar with this sort of platform development, and I'm a bit overwhelmed by all of the complexity. My goal is to get a Godot game running on the RG350; what are the first steps I need to take?

I'm very excited to learn more about this and hopefully contribute. Thank you so much!! 😄

efornara, 2020-11-12:
I just had a look at the reddit post you linked, and it looks like @ppiecuch has just posted a reply there asking for more feedback. I assume you own the hardware, so maybe you can help that way.

If you want to be more involved, I suppose a good start would be to download the toolchain, pull the relevant repositories and try to recompile the fork yourself. Where you go from there depends on your skills.

I don't see the github "Issues" tab enabled in the fork. I wouldn't mind the interaction to happen here, if it is of any help.

I hope to start soon working on FRT 1.0.x and easier cross compiling will be its main focus, although I more interested in debian. If the GCW0 toolchain turns out to work, it should be possible to adapt any changes needed and include them in a future version of FRT.

About RG350, from the little that I have read about it, these seems to be the main risks to watch out for:

- MIPS: Unlike ARM that is routinely run on Android, I don't think anyone is using recent versions of Godot on MIPS, so the engine itself might have problems with it.
- Vivante: Not all, but most of what is running on RG350 seem to be emulators that might not push the OpenGL ES 2 drivers that much. The engine might use more features that are implemented.
- DRM/KSM: It might be worth compiling FRT without Godot (see the porting directory) to make sure that this works as expected.

rcoppy, 2020-11-13:
Awesome, thank you so much for the advice! Unfortunately my skillset is pretty rudimentary 😅 (I'm more of an abstracted languages programmer than a low-level hardware guy) but the learning is part of the fun, and I have a bit of time on my hands. 🙂 

If it does turn out that Godot on MIPS isn't a thing anymore, I may just wait til I can get my hands on the RG351, which does use ARM.

I'll keep you posted!

rcoppy, 2020-11-20:
Hi again @efornara !

So I've spent the morning re-reading through all of the documentation in the repo/doing adjacent research; 

If I'm understanding things correctly the MIPS problem is fairly intractable given it's so niche; but on the other hand the ARM chip looks like it almost might be supported by FRT out of the box, specifically by using the `frt_098_323_arm64` binary over on sourceforge. 

Preliminarily, using the arm64 binary, I did get a demo project to compile! I don't yet have the actual hardware to test on, though. (I think I have an RK3399 Rockboard on hand somewhere, I'll see if I can dig it out.)

I still have some (possibly silly) outstanding questions: 

Could you say more about this section from 'Exporting a game'? 

> You can generally use the Linux X11 Platform as for Godot 2. However, if you use compressed textures (e.g. 3D Games), you need to select the appropriate texture format (Etc). Since this combination (Etc for X11) might not be well tested upstream, you might need to use the Android export template and generate a PCK/ZIP file instead.

Is this saying, use Android, but instead of exporting an APK, generate a PCK, which can be run as an app generically on ARM devices that use an OS other than Android?

Additionally, regarding PCK files: after the Linux/X11 export using the custom binary completes, I'm left with two outputs from the build: the game binary (sized at 75MB), but also a PCK (at a much smaller 3MB). Does the PCK do anything, or can I just run the binary by itself?

Finally: 

_If I'm reading the docs correctly_ (and several times I've gone through them and realized that I haven't, so maybe this is another one of those cases): 

FRT for Pi targets by default the VC4 graphics driver, using the X11 display API. 

The RK3326 uses a Mali Bifrost GPU, whose open-source driver is called [Panfrost](https://gitlab.freedesktop.org/panfrost), which as of September is actually [officially supported by ARM](https://www.cnx-software.com/2020/09/18/arm-officially-supports-panfrost-open-source-mali-gpu-driver-development/); here are [open-source Bifrost kernel drivers.](https://developer.arm.com/tools-and-software/graphics-and-gaming/mali-drivers/bifrost-kernel)

To my knowledge, the Panfrost driver is included upstream in Mesa.

My question: Do I need to reconfigure FRT to use Panfrost in place of VC4, and can I switch targeting X11 to targeting KSM/DRM?  How complicated is that? 

Panfrost is (sort of?) the successor to Lima (Panfrost is to Bifrost what Lima is to Utgard), and it does look like supporting Utgard takes [exporting a custom value ](https://github.com/efornara/frt/blob/master/doc/Utgard.md) for `FRT_KMSDRM_DEVICE`.

The ['Environment.md' doc](https://github.com/efornara/frt/blob/master/doc/Environment.md) seems relevant but I don't quite know what to do with it.

Thank you so much for your help! 😊

rcoppy, 2020-11-21:
Update! Got a FRT-compiled build onto my ARM board (RockPro64, RK3399)--execution has been failing 😢; initially the error was `cannot execute binary file: Exec format error` because I messed up part of the build step; I think I've finally gotten it right, because now the error is `error while loading shared libraries: libasound.so.2: cannot open shared object file: No such file or directory`

In the export settings I left '64-Bit' checked and unchecked 'Debug' in the save dialog.

A little bit of googling suggests this isn't a problem with FRT, but with the fact that a sound library Godot needs [isn't necessarily installed on Ubuntu by default.](https://askubuntu.com/questions/482478/libasound-so-2-cannot-open-shared-object-file-no-such-file-or-directory) 

My SBC doesn't actually have an internet connection (I've got it connected to my dev laptop via ethernet cable, and haven't quite figured out how to forward DNS resolution yet), but once I figure that out and get dependencies installed I'll see if the demo runs.

Cool stuff!

jlahman, 2020-11-25:
I'm also interested in using godot to make games for these rk3326 odroid go advance clones, and quickly tested the arm64 precompiled bin on my RK2020 using the 3.0 2d platformer demo.

Audio was choppy, controls didn't work(although that could be due to me using an image made for the OGA and not the RK2020), and the view was sideways. Seemed to work well besides that though.

Here's a screenshot for proof:

[image]

efornara, 2020-11-26:
@jlahman 

Audio problems might be the same as https://github.com/efornara/frt/issues/14. I will see if I can do a few tests some day soon based on the feedback from @charasyn 

Controls (i.e. joystick) in FRT are handled by code that is copied verbatim from upstream that should be fairly tested (but not necessarily on platforms like yours). Mapping should be handled by the Godot engine itself and the list of IDs is copied, IIRC, from the SDL2 codebase. I assume that from what you can see Emulation Station UI is handling the controls fine.

I have no idea about the screen orientation issue, but, as a first step to investigate it further, it would be helpful if you can figure out if FRT is using X11 or KMS/DRM. One way to check might be to force FRT to use one or the other using FRT_MODULES (see Environment.md)

jlahman, 2020-11-28:
Quick update: 

1. I found out that the device uses the kms/drm module, it would crash using the x11 one. Also, I forgot to mention that the devices panel is a vertical one. I don't know what the OS builds did to rotate the screen.
2. I got the controls to mostly work. In the script that I use to launch the game, I exported a custom SDL_CONTROLLERCONFIGDB* variable and most of the controls started working; except for the analog stick(in in-game menus). In my gameplay they work it seems.

So screen rotation and audio are the two main issues right now. @charasyn's  latest comment on #14 might have fixed the issue though.

*the config in case anybody else needs it (note: its a different sdl binding that what you should use for the OGA/OGA-BE as its for the rk2020): 
`export SDL_GAMECONTROLLERCONFIG="19000000010000000100000001010000,odroidgo2_joypad,platform:Linux,a:b1,b:b0,back:b10,dpdown:b7,dpleft:b8,dpright:b9,dpup:b6,leftshoulder:b4,lefttrigger:b12,leftx:a0,lefty:a1,rightshoulder:b5,righttrigger:b13,start:b15,x:b2,y:b3,"`

efornara, 2020-11-28:
I would consider pulling in libgo2 only as a last resort. It would be in any case an additional, optional FRT module separate from the other ones (bcm, x11, etc...). Something that should be feasible, but that looks a bit time consuming. It cannot really be done blindly either: someone with access to the real hardware should be developing the modules.

My preferred approach by far would be to try to extend the kmsdrm module instead. Assuming that I have got the library right, it looks like libgo2 is essentially a thin wrapper on top of DRM (and other stuff):

https://github.com/OtherCrashOverride/libgo2/blob/master/src/display.c

One option would be to play with rotation. libgo2 seems to use something like HAL_TRANSFORM_ROT_90, but I think DRM has a more generic rotation property that can be tried first. One option would be to extend kmsdrm to look for an environment variable like:

    FRT_KMSDRM_ROTATION=rotate-90

The generic method can be coded and tested on a Linux PC. I don't know if @peicou would be interested in doing it. If that doesn't work, adding HAL_TRANSFORM_ROT_90 should probably be attempted by someone with access to the real hardware.

This of course unless I am missing something and there is an easier way to do it.

jlahman, 2020-11-28:
That is indeed the correct library, apologies for not linking it originally. If it comes down to it and the generic rotation property doesn't work, I am willing to try to implement the HAL_TRANSFORM_ROT_90 in kmsdrm as I obviously have the hardware, however, it may take me a good while, as I haven't really ever done anything this low level or done any sort of drivers.

efornara, 2020-11-30:
@jlahman 

I plan to wait 1-2 weeks to give @peicou a chance to reply. Eventually, I am happy to give the generic method a try myself, but the time-frame in that case might be a few weeks. Even if it doesn't work on the RK2020, it makes sense to add it anyway, as it might work on some other platforms.

If you want to know if the generic method has any chance of working on the RK2020, I guess you could ask around or try to get hold of a version of modetest from libkms (or compile it yourself). It should be able to list all the properties that are exposed by the driver.

I don't know what you plan to do with Godot on the RK2020, but if you are writing your own games, depending on the complexity of the game, as a temporary workaround, you might be able to "fix" the viewport in the game code. OS.get_name() can be used to see if the game is running on the PC or on the console.

peicou, 2020-12-01:
My apologies for the late reply- I've been tied up with work lately. Feel
free to modify the module.
Thank you for keeping me in the loop!

rcoppy, 2020-12-01:
@jlahman Amazing that you got a demo running!! I've also been work-busy but it's fantastic to see your progress 😄

jlahman, 2020-12-05:
I compiled libkms on the RK2020 and after realizing that I had to stop any graphical processes for modetest to work, I got the following output running `./modetest -M rockchip`

I'm not sure if I'm missing anything, but I don't think the driver exposes the rotation property? Once again, I am very new to this so I could be wrong/looking in the wrong place.

```Encoders:
id	crtc	type	possible crtcs	possible clones	
65	60	DSI	0x00000001	0x00000000

Connectors:
id	encoder	status		name		size (mm)	modes	encoders
66	65	connected	DSI-1          	42x82		1	65
  modes:
	index name refresh (Hz) hdisp hss hse htot vdisp vss vse vtot)
  #0 320x480 60.02 320 450 454 584 480 482 483 485 17000 flags: nhsync, nvsync; type: preferred
  props:
	1 EDID:
		flags: immutable blob
		blobs:

		value:
	2 DPMS:
		flags: enum
		enums: On=0 Standby=1 Suspend=2 Off=3
		value: 0
	49 brightness:
		flags: range
		values: 0 100
		value: 50
	50 contrast:
		flags: range
		values: 0 100
		value: 50
	53 saturation:
		flags: range
		values: 0 100
		value: 50
	54 hue:
		flags: range
		values: 0 100
		value: 50

CRTCs:
id	fb	pos	size
60	69	(0,0)	(320x480)
  #0 320x480 60.02 320 450 454 584 480 482 483 485 17000 flags: nhsync, nvsync; type: preferred
  props:
	44 left margin:
		flags: range
		values: 0 100
		value: 100
	45 right margin:
		flags: range
		values: 0 100
		value: 100
	46 top margin:
		flags: range
		values: 0 100
		value: 100
	47 bottom margin:
		flags: range
		values: 0 100
		value: 100
	29 CABC_LUT:
		flags: blob
		blobs:

		value:
	28 CABC_MODE:
		flags: enum
		enums: Disable=0 Normal=1 LowPower=2 Userspace=3
		value: 0
	28 CABC_MODE:
		flags: enum
		enums: Disable=0 Normal=1 LowPower=2 Userspace=3
		value: 0
	57 FEATURE:
		flags: immutable bitmask
		values: afbdc=0x1
		value: 1

Planes:
id	crtc	fb	CRTC x,y	x,y	gamma size	possible crtcs
58	60	69	0,0		0,0	0       	0x00000001
  formats: XR24 AR24 XB24 AB24 RG24 BG24 RG16 BG16
  props:
	7 type:
		flags: immutable enum
		enums: Overlay=0 Primary=1 Cursor=2
		value: 1
	8 SHARE_ID:
		flags: immutable range
		values: 0 4294967295
		value: 58
	56 FEATURE:
		flags: immutable bitmask
		values: scale=0x1 alpha=0x2 hdr2sdr=0x4 sdr2hdr=0x8 afbdc=0x10 pdaf_pos=0x20
		value: 18
59	0	0	0,0		0,0	0       	0x00000001
  formats: XR24 AR24 XB24 AB24 RG24 BG24 RG16 BG16
  props:
	7 type:
		flags: immutable enum
		enums: Overlay=0 Primary=1 Cursor=2
		value: 2
	8 SHARE_ID:
		flags: immutable range
		values: 0 4294967295
		value: 59
	56 FEATURE:
		flags: immutable bitmask
		values: scale=0x1 alpha=0x2 hdr2sdr=0x4 sdr2hdr=0x8 afbdc=0x10 pdaf_pos=0x20
		value: 2
61	0	0	0,0		0,0	0       	0x00000001
  formats: XR24 AR24 XB24 AB24 RG24 BG24 RG16 BG16 NV12 NV16 NV24 NA12 NA16 NA24
  props:
	7 type:
		flags: immutable enum
		enums: Overlay=0 Primary=1 Cursor=2
		value: 0
	8 SHARE_ID:
		flags: immutable range
		values: 0 4294967295
		value: 61
	56 FEATURE:
		flags: immutable bitmask
		values: scale=0x1 alpha=0x2 hdr2sdr=0x4 sdr2hdr=0x8 afbdc=0x10 pdaf_pos=0x20
		value: 3

Frame buffers:
id	size	pitch
```

efornara, 2020-12-05:
I'm not an expert myself and I'll have to do a bit of reading before I change the kmsdrm module. Yes, it looks like it isn't exposed. To be honest, I don't know how widespread the support is. I guess I'll find out more when I compile modetest myself. Some drivers seem to support it, though. This seems to be how it should look like (https://www.linux4sam.org/bin/view/Linux4SAM/UsingAtmelDRMDriver):
```
...
Planes:
id   crtc   fb   CRTC x,y   x,y   gamma size   possible crtcs
11   15   31   0,0      0,0   0          0x00000001
  formats: XR12 AR12 RA12 AR15 RG16 RG24 XR24 AR24 RA24
  props:
   4 type:
      flags: immutable enum
      enums: Overlay=0 Primary=1 Cursor=2
      value: 1
   10 rotation:
      flags: bitmask
      values: rotate-0=0x1 rotate-90=0x2 rotate-180=0x4 rotate-270=0x8
      value: 1
...

```

I plan to add rotation to kmsdrm anyway, but it isn't likely to help you.

If you ever get to to modifing kmsdrm, here are a couple of remarks.

As you might have already seen, the key seems to be:

    #include <rga/RgaApi.h>

and related defines and functions. So, there might be a special library that you need to use. For a fork or for the initial development, statically linking it is OK, but for a PR, it has to be optionally compiled and dynamically loaded.

I am going to change a bit the code of the kmsdrm module (I think today or tomorrow, but I might need to reschedule). To avoid problems with merging, make sure to wait for an updated version before doing anything more than throwaway tests.

rcoppy, 2021-06-11:
Hi everybody! I know it's been a minute but this whole SBCs with Godot thing keeps ending up back in my head 😝 

In terms of Odroid-clone handheld devices--the market has changed once again! The RK2020 chipset has largely been superseded by the RK3326 / [the Odroid Go Super has replaced the OGA.](https://obscurehandhelds.com/2020/12/the-hardkernel-odroid-go-super-promises-big-improvements/)

The current generation of SBC handhelds are largely RK3326-based and running variants of EmuElec. 

I got in touch with one of the devs over on the EmuElec discord to ask about properly targeting Odroid-like hardware and he had this to say:

[image]

So basically as of June, 2021 in terms of FRT targeting SBC handhelds, the most prolific target architecture at the moment is *aarch64* (which confusingly is somewhat but not completely interchangeable with **arm64**), with *KSM/DRM* seemingly the best current option for display manager.

I don't yet have my hands on an RK3326 handheld (still just have my MIPS machine that started this whole thread off), but hopefully within a month I'll be able to do some playing around 🙂 

Really awesome to see that the audio bug (#14) ended up being fixed all the way upstream in the official godot branch, and that ARM device and cross-compilation support is a godot 4.0 milestone!

efornara, 2021-06-11:
I wouldn't read too much into aarch64 vs arm64 when it comes to FRT. I have used arm64 just because it is the official name for the ARM port on debian. For FRT 2.0, and some preview builds of FRT 1.x too if I decide to make them, I will abandon the pi1/pi2/arm64 naming and move to arm32v6/arm32v7/arm64v8.

As for Godot 4.0, the big question mark for me (and FRT) is whether OpenGL (3 and maybe 2) will be supported from the start, or whether it will be postponed to Godot 4.1. Related questions are what the state of the low-end Godot 4.0 Vulkan render path will be and how good the Vulkan drivers for common SBCs (i.e. PanVK) will be.

The fact that Fbdev is better on new hardware like OGA/OGS surprises me. I thought that KMS/DRM was supposed to replace it. Interesting.

rcoppy, 2021-06-11:
Interesting stuff about godot 4.0! 

A guy named Nils got in touch with me over email, pointed me to this forum thread (which @efornara was active in); just so other folks can find it I'll leave it here:

https://forum.odroid.com/viewtopic.php?f=193&t=39708

closed 2022-01-29 by efornara