efornara/frt #28: FRT 2.0

efornara, 2021-03-17:
You might have noticed a new 2.0 branch, so here is an explaination of what is happening. It's a long post, so here is an:

### Executive Summary

**FRT 2 will be a FRT rewrite using a patched SDL2 instead of custom code. For a long time, FRT 1 will still remain the main FRT version, and will continue to be patched to keep it working.**

Here are the details:

**Why rewriting FRT from scratch?**

It is not a big deal, really. For me, the value of FRT is in more in the experience that I have gained, than in the code itself, that I consider expendable and isn't a lot in the first place.

I am probably going to regret saying this, but I don't see why it should take more than a couple of weekends.

**When?**

That it might take me just a couple of weekends, doesn't say much about when they might happen. I might have a window in the second half of April. In May-June, I expect to patch SDL2 anyway for another project, so that is another option. But that fact is, I cannot say.

**Why SDL2?**

When FRT started, the BCM module was king. The X11 module was (re-)added as a way to make sure that, if you distributed your game with FRT, it would at least run in some fashion for the small minority of users who upgraded to the new, experimental VC4 driver or who had a board that had hardware-accelerated X11.

Times have changed. X11 and KMS/DRM are becoming more and more common. Functioning open source drivers for SBCs built on top of Mesa have started appearing. While desktop Wayland might be slow to reach mainstream, some people are already reporting that a lightweight Wayland composer gives them longer battery life on their laptops. I tested weston on a Mali 400 board and I can confirm that Wayland works and it is now at least a viable option.

It is cheaper to rewrite FRT on top of SDL2, than to add proper support for X11, KMS/DRM and Wayland to FRT with custom code.

**Why patching SDL2?**

Building against the system / standard SDL2 will be supported. I expect most developers will compile Godot/FRT that way.

I cannot do that. SDL2 doesn't really support the hardware scaler of the Pi 0-3, but it just ignores the passed window size:

https://github.com/libsdl-org/SDL/blob/release-2.0.14/src/video/raspberry/SDL_rpivideo.c#L279-L281

	/* Windows have one size for now */
	window->w = display->desktop_mode.w;
	window->h = display->desktop_mode.h;```

Also, looking at https://github.com/efornara/frt/issues/25 I would really like to patch the SDL2 KMS/DRM video driver to load librga at runtime.

Hopefully, my changes can be pushed upstream, but in any case I will find it faster to work on a fork of mine first.

From an user point of view, it will be transparent. SDL2 will be statically linked in the official FRT 2 binaries.

**What about Godot 4? Godot 2?**

One of the main reason why the FRT code is so hacky (I am looking at `os_frt.cpp` here with all its `#ifdef`s) is that I started tracking the Godot 3 code base too early. In highsight, waiting and doing a refactory would have been better, but, (un)fortunately, the patch-on-top-of-patch code ended up working.

I should occassionaly have a go at Godot 4, just for fun and to play around with Vulkan. However, I don't expect to really push for it until Godot 4 has reached the release candidate stage. Also, while I will do a best-effort attempt, I cannot guarantee that FRT will still be viable there. What I can say now, for the very little that I have seen, is that, at least at the beginning, FRT will be a custom SDL2 display server for a (minimally) patched Godot 4 and not a custom platform for an unpatched Godot 4.

Chances of FRT 2 not fully supporting Godot 2: zero.

**What about FRT 1?**

FRT 1 is here to stay. Even when I start releasing FRT 2 binaries, for a while the "big green download button" on sourceforce will point to FRT 1. And even when I switch it to point to FRT 2, I plan to keep releasing FRT 1 binaries for a few months still.

efornara, 2021-06-18:
How is FRT 2 doing?

At the moment I am "stuck" in a preliminary step (i.e. crossbuild). By "stuck" I don't mean that I have problems, but that the crossbuild that I have is good enough for what I need right now, but not good enough for FRT 2.

The next round of crossbuild changes might make it good enough for FRT 2 too, and, finger crossed, it might happen soon. Then, the next step is likely to be the recompilation of FRT 1 code with the FRT 2 toolchain.

Once that goes well, work on FRT 2 proper might start at any time, but, again, I don't know when.

Speaking of which, some of you might be aware that I did run a little experiment to see how much interest there was for FRT 2 in the community:

https://twitter.com/efornaralabs/status/1374327763587465216

Well, the experiment ended and the results are in: there is no really much interest for it.

What does it mean for FRT 2? Not much, really. I still plan to write FRT 2 sooner or later. It will just be a little later. The other minor difference is that I will start with Godot 2 (the version that I use) first, while if it the development was community-driven I would have started with Godot 3.

efornara, 2021-11-01:
@rcoppy

I took a break from both my game and FRT (I have only been releasing binaries, no new code). I plan to reengage with both very soon. I think my plan for FRT2 are still pretty much the same, just delayed. I might be able to have a go next weekend or the one after that.

Your help would be very appreciated. The main stumbling block for FRT2 was and still is the build environment. The problem is that the one I am using now (https://github.com/efornara/crossbuild) is good enough for my game, but not for FRT2. I expect issues with scons (needed by Godot) and clang (needed specifically by Godot 2.1). I do have a Pi 3B+, but my main SBC is an Orange Pi PC. It is 32 bits and I cannot easily generate 64 bit binaries from there. Initially, I plan to release armv7 (bullseye) binaries only. It would be very helpful if someone could build more useful binaries (e.g. 64 bits, buster, etc...) from the same commit, not necessarily from a cross environment.

Also, I was wondering. IIRC, when I checked EmuElec commits a while ago there was a rollback to a previous version of SDL2 (it was even before 2.0.16 was released). Do I remember right? How is the situation now? Does EmuElec use an old version of SDL2? If so, why?

rcoppy, 2021-11-03:
I'm not sure about the EmuElec stuff but I can definitely ask over in the discord!

shantigilbert, 2021-11-05:
> Also, I was wondering. IIRC, when I checked EmuElec commits a while ago there was a rollback to a previous version of SDL2 (it was even before 2.0.16 was released). Do I remember right? How is the situation now? Does EmuElec use an old version of SDL2? If so, why?

Yes, in fact SDL 2.0.9 is used in the Amlogic project, mostly because I am still using FBDEV, and the patch for MALI FBDEV was only ever released for 2.0.9, while I could port it to 2.0.12, it had video errors and 2.0.14 was a no go, I am not knowledgeable enough in SDL to make the port to SDL 2.0.14+ 

Reason I still use FBDEV is because I had many issues when I tried using DRM/GBM with the MALI blobs. 

And on the OdroidGoAdvance we use SDL 2.0.10 because SDL 2.0.14 was failing again with the MALI blobs creating the screen surface, but in any case this version is easy(or easier) to update to a more recent version granted there is a rotation patch available.

rcoppy, 2021-11-05:
Thanks for the quick response @shantigilbert ! 🙂

efornara, 2021-11-06:
Thanks @shantigilbert .

I also don't know SDL that well yet, but I should get to know it better with time. Improving the support for FBDEV in the latest version of SDL is something I am interested in, but not in the short term.

What I can do in the short term is to make sure that FRT can compile with older versions of SDL; maybe not for its first commit, but soon enough.

efornara, 2021-11-14:
I've just pushed the first FRT 2 commit. It is mainly an "internal" release, but now I can at least post a status update and a tentative roadmap.

This commit runs Godot2 on a PC and it only has initial support for video output and keyboard input. So, not really worth wasting time testing it. It's enough to play the 2d platformer, tough.

I haven't tried compiling it and running it on any SBC yet, but I don't expect problems there. I'm more interested in having a go at cross compiling, starting with arm32v7 and arm64v8 (buster). It might take a few days of low-intensity activity (i.e. change a compilation flags, do something else for 1-2 hours, test, repeat) to see if cross compiling is feasible.

After that, I would like to add audio and joystick support, but passing through SDL2. FRT 1 re-uses the upstream Godot drivers. This would give me, e.g., PipeWire support on every Godot version; and if a weird platform doesn't use udev, it would be possible to leverage a patched SDL2 to access its controls.

Finally, I would like to have a short try at supporting Godot 4. Even if I find blocking problems there, I hope that spending some time trying should better future proof any Godot2/Godot3 abstraction layer I end up writing.

Time estimate? I don't know, but I don't want to drag it too long, so I hope to have some sort of release candidate for wider testing by the end of the year.

efornara, 2021-12-02:
I've just released the first binaries of what I could call FRT 2.0 "alpha":

https://sourceforge.net/projects/frt/files/previews/

You cannot use it to try your own game with it (still Godot 2 only) and some (comparatively easy) stuff is still missing, but I reckon that most of the "critical" code is there: Video, Audio and Joysticks are all handled using SDL2 now.

At some point all features have been tested somewhere (most of the times on a PC, but occasionally on SBCs too), but I haven't done any systematic testing of the final binaries, so expect issues.

If you need Godot 2 projects, you can use demos from here:

https://github.com/godotengine/godot-demo-projects/tree/2.1

No need to export. I've used mainly:

misc/joysticks (no vibration)
misc/window_management (some bits still missing)
2d/platformer (seems ok)

Since these demos are usually designed to be exited by closing the window (something you might not have), I have added a couple of temporary shortcuts to exit FRT: Alt+Enter or Joystick Button 6; they can de disabled by setting the FRT_NO_EXIT_SHORTCUTS environment variable.

I plan to stay on Godot 2 for 1-2 weeks still, but please, if you can, don't wait for Godot 3 to do some testing: most bugs are more quickly and easily dealt with in Godot 2.

If you do compile your own version of FRT, use clang if you can. If you only have gcc available and it is a recent one (IIRC, 6+), compile with target=debug or you'll have random crashes (known Godot 2 issue).

efornara, 2021-12-19:
I've just released the binaries of what I could call FRT 2.0 "beta":

https://sourceforge.net/projects/frt/files/previews/

There are a few improvements compared to "alpha", but the main difference is that this version supports Godot 3 (at least 3.4.1-stable, older versions like 3.0.6 might need minor tweaks).

The same caveats about lack of systematic testing apply, so expect issues.

What is needed the most now is probably some outside testing. This version being compiled against the far more popular Godot 3 and the holiday season giving people some spare time for hobbies will hopefully help.

There are plenty of little things still to do, but the best I can do for a little while is probably test some more and write some documentation / code comments.

efornara, 2022-01-23:
I went ahead and released FRT 2.0.0. No point in waiting. The code didn't change much, so still plenty of little things missing, but they can wait 2.0.1. I did some testing and I've found it usable.

The main difference compared to the "beta" (and it is a big one) is that I've decided to publish binaries with SDL2 dynamically linked. It happen to be SDL2 2.0.9 (the one in buster), but I've used my fork of 2.0.16 to override it and I had no issues.

efornara, 2022-01-29:
Discussion moved to https://github.com/efornara/frt/issues/39.

closed 2022-01-29 by efornara