efornara/frt #30: Screen Rotated 90 Degrees on RG351

desttinghim, 2021-06-14:
This has been mentioned in #25,  but the screen is out 90 degrees when running on the RG351. It looks like a way to fix this was proposed, but it doesn't seem like it was implemented, or at least not documented. I've just started using FRT and was surprised by how easy it was to get working. I'd like to help resolve the issue with the screen having the wrong orientation.

efornara, 2021-06-14:
**Generic KMS/DRM**

As for the generic / standard path (i.e. pure KMS/DRM), in that thread I had this to say:

> To be honest, I don't know how widespread the support is. I guess I'll find out more when I compile modetest myself.

> I plan to add rotation to kmsdrm anyway, but it isn't likely to help you.

I did run modetest a few months ago and it looked like it wouldn't be useful for my boards: H3-based (a couple of Orange Pis) and VC4-based (Raspberry Pis before 4). RG351 doesn't seem to use it either. So, given its poor support, I do *NOT* plan to add it to FRT1.

One of these days I should check if the drivers situation got better: I would really love for KMS/DRM scaling to be supported at least on VC4. If so, I might add it, but in any case to my fork of SDL2 (i.e. FRT2). 

**RG351**

@desttinghim 

At the moment, the easy path is to just rotate the viewport in your game if you can. IIRC, how to do it is described in the PDF linked here https://forum.odroid.com/viewtopic.php?f=193&t=39708 Not ideal, I know.

You are welcome to help, but that means pretty much doing most of the work yourself. I don't have the hardware, and this is the kind of work that usually requires quite a few iterations. It cannot really be done blind. Tomorrow, I might add a couple of devices to my wishlist on amazon/twitter, but I wouldn't hold much hope for any kind soul appearing ;-)

You would need to be fairly familiar with C(/C++) and the Linux C toolchain. Here is the relevant bit from the other thread:

> If you ever get to to modifing kmsdrm, here are a couple of remarks.
>
> As you might have already seen, the key seems to be:
>
>    #include <rga/RgaApi.h>
>
> and related defines and functions. So, there might be a special library that you need to use. For a fork or for the initial development, statically linking it is OK, but for a PR, it has to be optionally compiled and dynamically loaded.

Keep in mind that any work you do won't automatically apply to SDL2/FRT2. Still, any knowledge gained would be very valuable, and should make it fairly easy to port it there later if you want. Or you could just use SDL_DYNAMIC_API...

desttinghim, 2021-06-15:
I've made a tetris clone using SDL2 that my brother ported to the RG351. There was no issue with rotation with that game, so I think FRT2 being based on SDL2 would prevent the issue.

rcoppy, 2021-06-16:
Re FRT2/SDL2--how is development going? If you ever need any beta testers I'd be happy to help 🙂 

I should have access to RK3326 hardware within the next month (probably a gameforce chi).

efornara, 2021-06-18:
Thanks @rcoppy I haven't tried yet, and it might turn out to be unfeasible, but at the next round of crossbuild changes, I would like to have a go at RG350 / JZ4770 / OpenDingux too and some testing would be very helpful, assuming you are not planning to sell it.

Cebion, 2021-07-16:
Hey one of the devs of 351ELEC here.
I do have a RG351P/V on hands and if you're ready I'd also like to test it.
Awesome work! 

This would also help with the invisible cursor and make so much more games playable.

efornara, 2022-01-29:
Closing all FRT 1 issues. Please comment on https://github.com/efornara/frt/issues/39 if you get a chance to try again with the latest 2.0 binaries, and the problem is still there.

closed 2022-01-29 by efornara