efornara/frt #31: Trying to get FRT running on handheld RG351V with EmuElec OS

nilsnordmark, 2021-06-14:
Hi,
Trying to run my Exported Godot games on the RG351V is and Odroid Go Advance clone.
Specs: RockChip RK3326, Cortex-A35, Mali-G31 MP2, 1GB Memory.

I've followed this Guide mostly: https://forum.odroid.com/viewtopic.php?f=193&t=39708
I've seen some got it working with the 'Stock' OS. Alltough mine came with EmuElec V 2.8

I've tried with all three FRT options, arm64/pi1/p2 without success.

```
RG351:~/roms/godot # ./frt_100_332_pi1.bin leoparden_rg351v_pi1.pck
frt: dlopen failed for 'libgbm.so.1'.
frt: no video module available.
RG351:~/roms/godot # ./frt_100_332_pi1.bin leoparden_rg351v_pi2.pck
frt: dlopen failed for 'libgbm.so.1'.
frt: no video module available.

```

```
RG351:~/roms/godot # ldd frt_100_332_arm64.bin
        not a dynamic executable
RG351:~/roms/godot # ldd frt_100_332_pi1.bin
        /usr/lib/libarmmem-v7l.so (0xf756b000)
        libz.so.1 => /usr/lib/libz.so.1 (0xf7558000)
        libpthread.so.0 => /usr/lib/libpthread.so.0 (0xf752f000)
        libdl.so.2 => /usr/lib/libdl.so.2 (0xf751c000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0xf73d7000)
        libm.so.6 => /usr/lib/libm.so.6 (0xf736d000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0xf7340000)
        libc.so.6 => /usr/lib/libc.so.6 (0xf7206000)
        /lib/ld-linux-armhf.so.3 => /usr/lib/ld-linux-armhf.so.3 (0xf7580000)
RG351:~/roms/godot # ldd frt_100_332_pi2.bin
        /usr/lib/libarmmem-v7l.so (0xf73c5000)
        libz.so.1 => /usr/lib/libz.so.1 (0xf73b2000)
        libpthread.so.0 => /usr/lib/libpthread.so.0 (0xf7389000)
        libdl.so.2 => /usr/lib/libdl.so.2 (0xf7376000)
        libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0xf7231000)
        libm.so.6 => /usr/lib/libm.so.6 (0xf71c7000)
        libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0xf719a000)
        libc.so.6 => /usr/lib/libc.so.6 (0xf7060000)
        /lib/ld-linux-armhf.so.3 => /usr/lib/ld-linux-armhf.so.3 (0xf73da000)
RG351:~/roms/godot #
```

Any pointers would be much appreciated. 
Thanks
Best regards,
Nils

efornara, 2021-06-14:
It looks like the distro is 32 bits and the CPU is a Cortex, so frt_100_332_pi2.bin should be the best one, but if you manage to make it work and it crashes randomly, you could try to downgrade to frt_100_332_pi1.bin later.

Missing libgbm.so.1 could mean that the distro is not using KMS/DRM. My guess is that it could be using the proprietary Mali blob and not the new open source drivers.

FRT includes an fbdev module. It theory it supports Mali cards, but I have only tested it on an old card (A10 / Mali 400), so I am not sure if anything changed since, and it is years since I've last tested it anyway. Still, the fbdev module should be compiled in, just not auto-detected.

You can try this before starting your game:

    export FRT_MODULES=video_fbdev,keyboard_linux_input,

Also, it is probably better to set the proper screen size in the project settings, or to override it on the command line.

nilsnordmark, 2021-06-15:
Thanks for the quick response. I guess these things are very distro dependant. 
I've used the export-command you gave and FRT gives me this:

```
RG351:~/roms/godot # ./frt_100_332_pi2.bin --main-pack leoparden_rg351v_pi2_32bit.pck --verbose
frt: eglGetDisplay failed.
RG351:~/roms/godot # ./frt_100_332_pi1.bin --main-pack leoparden_rg351v_pi1.pck --verbose
frt: eglGetDisplay failed.
```

Tried setting resolution/GLES2/GLES3 driver etc 

I'm connecting to the handheld from my PC via SSH when running this. I've killed Emulationstation on the RG351V (EmuElec) and now it's just displaying the ANBERNIC-logo. I was expecting the terminal to show instead. 
Last time I did this was on the Rpi Zero + GPi Case and got it working there. 
My problems could be because Display is already busy?

efornara, 2021-06-15:
Starting FRT via SSH might be a problem. I know it is time consuming, but the next test I would do is to add an emulator script and let Emulationstation start FRT, ideally redirecting 2> and 1> somewhere accessible and known, so you can later check them via SSH.

IIRC, Emulationstation is designed to have emulators run as separate processes, so, assuming display busy is the problem here (I don't know), I expect Emulationstation to release it before starting FRT.

As for terminal vs logo, I personally would have expected a blank display instead of a full-fledged console login screen, but a logo doesn't worry me too much: it could be the firmware's idea of what a blank display should look like.

nilsnordmark, 2021-06-17:
Launching from Emulation Station seems to cause the same issue:

```
RG351:/emuelec/logs # cat es_launch_stderr.log
frt: eglGetDisplay failed.

RG351:/emuelec/logs # tail es_log.txt

2021-06-17 22:49:00     INFO    Configure emulators command : -p1index 0 -p1guid 03000000091200000031000011010000
2021-06-17 22:49:01     INFO            /storage/roms/godot/leo.sh
2021-06-17 22:49:02     WARNING ...launch terminated with nonzero exit code 256!
2021-06-17 22:49:02     INFO    Window::init
2021-06-17 22:49:02     INFO    Creating window...
2021-06-17 22:49:02     INFO    Created window successfully.
2021-06-17 22:49:02     INFO    Checking available OpenGL extensions...
2021-06-17 22:49:02     INFO     ARB_texture_non_power_of_two: MISSING
2021-06-17 22:49:02     INFO    Added known joystick OpenSimHardware OSH PB Controller (instance ID: 1, device index: 0)
2021-06-17 22:49:02     INFO    SDL AUDIO Initialized
2021-06-17 22:49:02     WARNING ...Show Error message! exit code 256!
2021-06-17 22:49:02     INFO    SDL AUDIO Deinitialized
2021-06-17 22:49:13     INFO    Window::init
2021-06-17 22:49:13     INFO    Creating window...
2021-06-17 22:49:13     INFO    Created window successfully.
2021-06-17 22:49:13     INFO    Checking available OpenGL extensions...
2021-06-17 22:49:13     INFO     ARB_texture_non_power_of_two: MISSING
2021-06-17 22:49:13     INFO    Added known joystick OpenSimHardware OSH PB Controller (instance ID: 2, device index: 0)
2021-06-17 22:49:14     INFO    SDL AUDIO Initialized
2021-06-17 22:49:14     INFO    Added known joystick OpenSimHardware OSH PB Controller (instance ID: 2, device index: 0)

RG351:~/roms/godot # cat leo.sh
#! /bin/sh
export FRT_MODULES=video_fbdev,keyboard_linux_input,
/storage/roms/godot/frt_100_332_pi2.bin --main-pack /storage/roms/godot/leoparden_rg351v_pi2_32bit.pck --verbose
```

Error 256 seems to be some no display error, when I google around.

In the es_systems.cfg in can see most emulators use emuelecRunEmu.sh. While "ports" don't.

```
<system>
    <name>3do</name>
    <fullname>Panasonic 3DO</fullname>
    <manufacturer>Panasonic</manufacturer>
    <release>1993</release>
    <hardware>console</hardware>
    <path>/storage/roms/3do</path>
    <extension>.iso .ISO .bin .BIN .chd .CHD .cue .CUE .zip .ZIP .7z .7Z</extension>
    <command>/emuelec/scripts/emuelecRunEmu.sh %ROM% -P%SYSTEM% --core=%CORE% --emulator=%EMULATOR% --controllers="%CONTROLLERSCONFIG%"</command>
    <platform>3do</platform>
    <theme>3do</theme>
    <emulators>
      <emulator name="libretro">
        <cores>
          <core default="true">opera</core>
        </cores>
      </emulator>
    </emulators>
  </system>
  <system>
    <name>ports</name>
    <fullname>Ports</fullname>
    <manufacturer>Ports</manufacturer>
    <release>Varies</release>
    <hardware>port</hardware>
    <path>/storage/.config/emuelec/ports</path>
    <extension>.sh .SH</extension>
    <command>/usr/bin/bash %ROM% -P%SYSTEM% --core=%CORE% --emulator=%EMULATOR% --controllers="%CONTROLLERSCONFIG%"</command>
    <platform>ports</platform>
    <theme>ports</theme>
[... 122 lines cut ...]
        echo 30 10 669 469 > /sys/class/graphics/fb0/window_axis
        echo 640 > /sys/class/graphics/fb0/scale_width
        echo 480 > /sys/class/graphics/fb0/scale_height
        echo 0x10001 > /sys/class/graphics/fb0/free_scale
        ;;
"576cvbs")
        fbset -fb /dev/fb0 -g 1280 960 1280 1920 $BPP
        fbset -fb /dev/fb1 -g $BPP $BPP $BPP $BPP $BPP
        echo 0 0 1279 959 > /sys/class/graphics/fb0/free_scale_axis
        echo 35 20 680 565 > /sys/class/graphics/fb0/window_axis
        echo 720 > /sys/class/graphics/fb0/scale_width
        echo 576 > /sys/class/graphics/fb0/scale_height
        echo 0x10001 > /sys/class/graphics/fb0/free_scale
        ;;
esac

fi
# End of reading the video output mode and setting it for emuelec to avoid video flicking.
# The codes can be simplified with "elseif" sentences.
# The codes for 480I and 576I are adjusted to avoid overscan.
# Forece 720p50hz to 720p60hz and 1080i/p60hz to 1080i/p60hz since 50hz would make video very choppy.

```

RG351V has a 640x480 panel

efornara, 2021-06-18:
Thanks for the test.

I suppose as an additional test, you could try to change leo.sh to something like this:

    #! /bin/sh
    
    if [ -e /proc/device-tree/t82x@d00c0000/compatible ]; then
      /emuelec/scripts/setres.sh 16
    elif [ -f "/emuelec/bin/fbfix" ]; then
      /emuelec/bin/fbfix
    fi
    
    export FRT_MODULES=video_fbdev,keyboard_linux_input,
    /storage/roms/godot/frt_100_332_pi2.bin --main-pack /storage/roms/godot/leoparden_rg351v_pi2_32bit.pck --verbose

It would also be helpful to know if SSH is an additional problem or is a non-issue, so killing and restarting Emulationstation via SSH is the last test I can think of.

But I don't really know what the problem is.

I am having a look at ELEC. I couldn't find 2.8, so I downloaded 2.7.1 from https://github.com/EmuELEC/EmuELEC/releases/tag/v2.7.1

It looks like at least /usr/bin/retroarch in there is linking against SDL, so it would be useful to track down where the SDL library comes from, so I can check if it has been patched. The image I have has "Amlogic" in its name, so I wouldn't assume it is the same for a RK3326 device. Is there a place where I can download the image you are using? Like the support page of the device you have? In theory, since there is a lot of GPL code in there, it should also have a link to the source code.

One thing that I have tried is to run my crossbuild/hello (i.e. what FRT2 would be) under qemu/chroot setting SDL_DYNAMIC_API to point to the SDL2 included in the image. Unfortunally, it doesn't work. crossbuild/hello uses a (patched, but it shouldn't matter) 2.0.14, while in the image it looks like a (patched?) 2.0.9 is used, and the SDL_DYNAMIC_API override fails.

nilsnordmark, 2021-06-18:
Thank you for your generous support!
I realise that the device RG351V is running a fork of EmuElec: https://github.com/351ELEC/351ELEC
My device says 2.8 release, but looking at the github I can't figure out the versioning. I will update to the latest they have there.

I'll try what suggested in previous post and get back to you!

efornara, 2021-06-18:
It might not be worth risking an update. The info that I need (what changes they are doing to the stock SDL2) is probably there. In the coming days I'll have a look.

nilsnordmark, 2021-06-19:
Updated leo.sh didn't change anything and neither did stopping Emulation station before running.
There is also an alternative OS that is popular called ArkOS, (for the RG351x handhelds):
https://github.com/christianhaitian/arkos

nilsnordmark, 2021-06-20:
Retrogamescorps just released a guide on how to get Pico8 running on RG351x devices with for example Emuelec or 351Elec. May be something there. I will look into it 

https://retrogamecorps.com/2020/11/12/guide-pico-8-on-retro-handhelds/

nilsnordmark, 2021-06-20:
Good news! So turns out i've been running EmuElec (stock) all along. I've switched to the fork 351elec and this is a 64bit distro so the FRT 64bit template worked right out of the box from ssh-command line. The rotation of the screen in the Godot that the Odroid GO Advance guide mentions is never necessary. The Godot export should work without tinkering. My game was not fullscreen and sound is really bad. Will do more testing and post a video + post mortem soon.

nilsnordmark, 2021-06-20:
Fixed both sound and resolution issue. I've set the resolution by FRT parameter --resolution 640x480. And sound issue was fixed by solution in another thread here: In Godot Project Settings -> Audio-> Mixrate from 44100 to 48000 and Output latency from 15 to 20 ms.

Video of final result: https://www.youtube.com/watch?v=NliQMAVCZWk
Hard to see on the video, but the screen of the RG351V is super sharp. Much much better than the Retroflag GPI Case+ Rpi Zero which is previously used (looked grainy).

Small Guide to get the FRT Godot export working on the RG351x:

1. Upgrade from stock OS to 351Elec https://github.com/351ELEC/351ELEC/
2. Download FRT export-template for 64 bit arm. https://github.com/efornara/frt/releases
3. Change Audio Settings Mixrate to 48000 and Latency to 20 ms.
4. Do not do anything crazy with screen rotation.
5. Export your game with FRT template selected and 64 bit, nothing else ('runnable' can be used)
6. Run from ssh-command-line or setup a Godot section in Emulation Station (there are guides out there). If running from command line you have to stop Emulation station by using the command: /usr/bin/systemctl stop emustation.service 
This is not necessary when setting up to run from Emulation Station UI.
7. Could be usefull to add --resolution parameter for settings correct resolution.

closed 2021-06-21 by nilsnordmark