efornara/frt #32: Using GDNative with frt?

SebastianBoyd, 2021-09-01:


efornara, 2021-09-02:
I haven't tried personally, but here someone got it working:

https://labs.ullo.fr/tutorials/python-and-godot-engine-on-a-raspberry-pi-displaying-physiological-signals-with-lsl/

It looks like they had to edit the .gdnlib file directly, adding an FRT entry:

> **Importing custom godot-python to the project**
> [...]
> Because FRT declares a new platform upon compilation, we have to tune pythonscript.gdnlib so it can link our version of the Godot “interpreter” with our version of godot-python. Edit the file and under the section [entry] add the line FRT="res://addons/pythonscript/x11-64/libpythonscript.so" (you can leave the other entries).

SebastianBoyd, 2021-09-07:
Thanks for the resource! Ill see if that works

closed 2021-09-07 by SebastianBoyd