efornara/frt #33: no /dev/dri/card found.

Meonix, 2021-09-03:
I'm starting build my server game and Bring it to raspberry pi 4 (Ubuntu Server OS) , I'm using this library to run my pck file but I got this error when I running with this command:
./frt_100_333_arm64.bin --main-pack mygameserver.pck
And I got this :
Godot Engine v3.3.3.stable.custom_build.b973f997f - https://godotengine.org
frt: no /dev/dri/card found.
I think my PCK file is not compile for headless server . So can you help me this problem , thank you so much.

efornara, 2021-09-03:
If I understand what you are trying to do, your best option is to use a headless or server build of the official Godot instead of FRT. I've never tried them myself, but the binaries that you find here:

https://github.com/hiulit/Unofficial-Godot-Engine-Raspberry-Pi

should work.

FRT expects some sort of display. I am not familiar with the distribution that you are using, but, given its name (Ubuntu **Server** OS), I am not that surprised if you run into problems.

Meonix, 2021-09-03:
thanks for your help but I tried with these lib but it still get error as : no such file or directory 
This error come out when I use headless server binary file of Unofficial-Godot-Engine-Raspberry-Pi library like this :
chmod +x ./godot_3.3.2-stable.rpi4+headless_lto.bin 
/godot_3.3.2-stable.rpi4+headless_lto.bin --main-pack mygameserver.pck
Error : no such file or directory

irishgreencitrus, 2021-09-04:
> thanks for your help but I tried with these lib but it still get error as : no such file or directory
>
> This error come out when I use headless server binary file of [...]

Hey there. You might want to look at my 64bit builds as Ubuntu Server doesn't have 32bit user space. https://drive.google.com/drive/folders/1Bd9-G7sNE6hr5IBo8Peh_iFVqU_sIjZ1

closed 2021-09-05 by Meonix

Meonix, 2021-09-05:
> Hey there. You might want to look at my 64bit builds as Ubuntu Server doesn't have 32bit user space. https://drive.google.com/drive/folders/1Bd9-G7sNE6hr5IBo8Peh_iFVqU_sIjZ1

  omg Thanks you so much .. You  saved me ! thanks youuuu <3

Meonix, 2021-09-06:
> Hey there. You might want to look at my 64bit builds as Ubuntu Server doesn't have 32bit user space. https://drive.google.com/drive/folders/1Bd9-G7sNE6hr5IBo8Peh_iFVqU_sIjZ1

but Can you tell me more about how to use it?

irishgreencitrus, 2021-09-06:
yeah that's how you use it. if you have more questions i'm on discord as Lime#1569