efornara/frt #37: No axes joysticks input from anbernic rg351p

Rel-Oiv, 2022-01-13:
Hi,

I have an anbernic rg351p, running 351elec (20211122 Pineapple Forest)
I have managed to get a demo working (godot 3.4.2 stable), both 2d and 3d (tried only gles2; 32-bit, had to rotate viewport) with audio, using FRT1.0 arm64. Hooray.

I also ported the joysticks demo project from the github repo mentioned to test the buttons. All the buttons work right, except the sticks. While the push on stick buttons work, the stick axes do not. Not even during the remapping (see video at the end).

I saw that FRT2.0 is worked on and wanted to bring this to attention. I don't have an arm system where to build FRT 2.0 template and to test screen rotation and sticks. I also wonder if this is a godot thing, that it doesn't understand the joysticks axes input from 351p.

Here is a video with the FRT1.0 exports (osTools and joystick demo)
https://user-images.githubusercontent.com/67479592/149308525-a2dc686f-1e68-4363-a795-6441c70d551c.mp4

efornara, 2022-01-15:
@Rel-Oiv 

> I have managed to get a demo working (godot 3.4.2 stable), both 2d and 3d (tried only gles2; 32-bit, had to rotate viewport) with audio, using FRT1.0 arm64. Hooray.

Great!

 > I also wonder if this is a godot thing, that it doesn't understand the joysticks axes input from 351p.

FRT 1 literally includes the upstream handling of joysticks as it is:

https://github.com/efornara/frt/blob/1.0.0/include/joypad_linux.cpp 

That code should be tested enough, so it should be possible to make it work.

I don't know about distributions and hardware revisions, here: https://old.reddit.com/r/godot/comments/rnoi0v/working_on_a_3d_platformer_for_the_rg351_handheld/ someone complains about the left analog stick input coming in backwards for a device that looks close enough to yours to me.

To try to make it work, you could play with SDL_CONTROLLERCONFIGDB. It might work, but it could be time consuming to find the right values. If you are lucky, you might be able to find the string for your device somewhere on the net. See https://github.com/efornara/frt/issues/25#issuecomment-735016844 for more info. By the way, you should be able to fix any inverted axis issues that way.

I don't think it is relevant, but I have to mention that I have a USB controller that starts in digital mode, and I have to press a button to switch to analog mode to be able to get its axis values. Even if you don't have a physical switch, if the SDL_CONTROLLERCONFIGDB route fails, as a last resort, you might want to look for a configuration setting about that somewhere.

Rel-Oiv, 2022-01-16:
Hello,

I have found inside 351elec a file called gamecontrollerdb, containing lots of sdl2 controller mappings, including 03000000091200000031000011010000 OpenSimHardware OSH PB Controller (which is the same with what godot returns from get_joy_guid() and get_joy_name(). With those settings, the buttons mapping was corrected, but still no input form sticks. I tried up to axis a15, with all devices id, but with no success.

I tried to see what retro arch uses, and found out that input controller is broken when is set on sdl2 instead of udev . No button works except the left stick in reversed (makes no sense but waterer), I gave up searching in retro arch, there is little to no info.

I tried a godot 3.2 export, but same results, no input joysticks axes input.
I tried to export the 2.1 joystick demo from godot version 2.1, but it crashes at boots.

I might ask a dev of 351elec about the input, maybe they have a clue.

Anyhow, I'll give up on the joysticks for now. Thanks for the hard work.

the joypad config sdl2 from 351elec
03000000091200000031000011010000,OpenSimHardware OSH PB Controller,a:b0,b:b1,y:b3,x:b2,start:b6,back:b7,dpup:h0.1,dpleft:h0.8,dpdown:h0.4,dpright:h0.2,leftshoulder:b4,lefttrigger:b10,rightshoulder:b5,righttrigger:b11,leftstick:b8,guide:b9,leftx:a0~,lefty:a1~,rightx:a2,righty:a3,platform:Linux,

efornara, 2022-01-18:
@Rel-Oiv 

> I'll give up on the joysticks for now.

Sorry it didn't work out. If later on you want to have another go, I agree that asking for some pointers to someone familiar with the device / distro is your best bet. Maybe it's a known issue and you are expected to do something out of the ordinary code-wise to get it working.

> I tried to export the 2.1 joystick demo from godot version 2.1, but it crashes at boots.

Thanks. It's been ages since I've last tried a release on a arm64 distro. Next chance I get, I'll have a look.

efornara, 2022-01-29:
Closing all FRT 1 issues. Please comment on https://github.com/efornara/frt/issues/39 if you get a chance to try again with the latest 2.0 binaries, and the problem is still there.

closed 2022-01-29 by efornara