efornara/frt #39: ToDo List

efornara, 2022-01-29:
Known issues with FRT 2.0.x. Kept up to date.

High Priority

- [ ] Web site
- [ ] Some more documentation
- [ ] Feedback without github registration (anonymous google form?)

Medium Priority

- [ ] Joystick vibra
- [ ] Audio supporting more than 2 channels
- [ ] Icon / custom cursor shape
- [ ] Common detect*.py funcs / classes
- [ ] GLES2 video driver fallback
- [ ] can_draw
- [ ] Godot 4 hello world (compile empty main or maybe minimal vulkan context)

Low Priority

- [ ] Switch master to 2.0 branch
- [ ] Batch compile on every stable release
- [ ] Find breaking points (correct FRT_GODOT_VERSION in ifdefs)
- [ ] arm32v6 docker image (chroot+qemu or sysroot?)
- [ ] Desktop-like features (copy and paste...)
- [ ] Godot 4 OS (events...)

Rel-Oiv, 2022-01-29:
Hello,
I tried a few 2D demos (ui nodes only) I did/ got from demo projects exported with FRT 2.0 binaries (I tried only frt_200_342_arm64v8.bin, with 64 unchecked in the godot's exporter) and the joysticks worked. Also the screen orientation was correct without any viewport rotation. Congrats for the release of FRT 2.0

Cebion, 2022-01-29:
I also tested with quite a lot of titles! 
Works really really well.
In combination with something like gptokeyb for mouse emulation and keyboard presses it will change the handheld market =)

rcoppy, 2022-01-29:
Really fantastic work!! Congrats on the release 🙂

zapposh, 2022-08-01:
Hi,
Just checking if Blaze / Evercade has been in touch with you.
We have managed to get FRT builds of the game to run on the Evercade console using arm32v7, but apparently the system specific optimization flags are not set, leading to poor performance. Would it be conceivable to implement these for a future update, with guidance from the team at Blaze, so that Godot users can potentially port their games over to Evercade?

efornara, 2022-08-02:
@zapposh I was mentioned in a tweet suggesting some DM exchanges. I followed their account for a few weeks, but I don't think I received anything. Anyway, I have now re-followed them again. If they prefer to have a public discussion here instead, posting the actual system specific optimization flags together with some statistics would be helpful.

zapposh, 2022-08-02:
@efornara Thanks for your reply. I'll see what I can do and get back to you.

zapposh, 2022-08-02:
ps.: your Twitter DMs seem to be closed, which might be why you did not get any further input from Blaze.

BlazeEntertainment, 2022-08-06:
Thanks @zapposh, sorry I have been away for a couple weeks. I dm'd you @efornara!

MTareqAzim, 2023-01-08:
Hello,
**Bug Report**
Joy axes trigger no events and axes mapped to actions are never updated. Polling the joy axes works regularly though.
0. Tested on an RG353M, using the frt_201_351 binary, on ArkOS 2.0, on arm64v8 architecture. Theoretically, this should affect every frt_201_351 binary.

In the file `frt_godot3.cc`, there is this function on lines 379 to 386:
```
void handle_js_axis_event(int id, int axis, float value) FRT_OVERRIDE {
#if FRT_GODOT_VERSION >= 30500
		input_->set_joy_axis(id, axis, value);
#else
		InputDefault::JoyAxis v = { -1, value };
		input_->joy_axis(id, axis, v);
#endif
	}
```
Specifically on line 381, are you sure you don't mean to be using `input_->joy_axis(id, axis, value);`?
The way it is set up right now completely bypasses the event system, thus also bypassing InputMap thus unable to update actions using joystick axes mapped to them.
Recompiled with the `joy_axis()` function instead and the actions are properly updated and joy axes trigger events now.

efornara, 2023-01-08:
@MTareqAzim

> are you sure you don't mean to be using input_->joy_axis(id, axis, value);?

Thanks. Most likely, yes that's what it should be. I cannot test it right now, but a quick look at x11 and it is in fact using joy_axis. I guess when upstream changed the method signature in 3.5, I mistakenly picked the wrong new one.

efornara, 2023-02-02:
Superseded by https://github.com/efornara/frt/issues/45

closed 2023-02-02 by efornara