efornara/frt #41: Missing builds on FRT 2.x

hiulit, 2022-11-18:
Hello there!

I've been trying to update my [RetroPie Godot Engine Emulator](https://github.com/hiulit/RetroPie-Godot-Engine-Emulator) to the latest FRT from `1.0.0` to `2.0.1` and some builds are gone:

- `3.1.2`
- `3.3.4`

There's a new one, `3.2.3`.

Why is that? I'm just curious. I thought you always compiled the latest minor version. That would be:

- `2.1.6`
- `3.0.6`
- `3.1.2`
- `3.2.3`
- `3.3.4`
- `3.4.5`
- `3.5.1`

Thank you!

efornara, 2022-11-18:
I don't release the binaries for every latest minor version anymore. The reason is too much time spent on compiling / uploading, and pretty much nobody downloading them.

My policy now is to follow upstream:

https://docs.godotengine.org/en/stable/about/release_policy.html

Someway arbitrarily, to that list I add 2.1.6 (might be useful for some boards), and versions that were ever in debian stable (but I might drop the oldest ones in the future).

With the docker scripts that I use to generate the binaries now public, it should be fairly easy to build whatever version you need. I still care about source-level compatibility of every release, so, if problems are reported, I try to reproduce and fix them.

hiulit, 2022-11-21:
Ok, thank you for your answer :)
It makes sense.

closed 2022-11-21 by hiulit