efornara/frt #42: Environment variables and extensions from 1.x missing

hiulit, 2022-11-21:
In `1.x` there were some environment variables https://github.com/efornara/frt/blob/1.x/doc/Environment.md.
They seem gone on `2.x`. Are they not relevant anymore?

Same for the extensions https://github.com/efornara/frt/blob/1.x/doc/Extensions.md

I was using some of them (`FRT_X11_UNDECORATED`, `FRT_KEYBOARD_ID`, `FRT_KMSDRM_DEVICE`) in my [RetroPie Godot Engine Emulator](https://github.com/hiulit/RetroPie-Godot-Engine-Emulator), but the `--frt exit_on_shiftenter=true"` was particularly useful.

Thank you!

efornara, 2022-11-21:
Exiting on shift+enter is the default right now, and you have to set an environment variable to disable it. I plan to reverse it though, ideally already in the next release.

FRT_X11_UNDECORATED shouldn't be needed, and the upstream command line option to force fullscreen should work (IIRC, -fs).

Selecting specific devices should now be done using the standard SDL environment variables.

hiulit, 2022-11-29:
Hi, thank you for you answer.

Indeed, using `-f` forces fullscreen.
I'll have a look for the devices.
What doesn't seem to work is the <kbd>shift + enter</kbd>.

efornara, 2022-11-30:
I've just checked the code, and indeed the current shortcut is alt+enter. The final one is likely to be shift+enter (IIRC, it is the default mapping on RetroPie).

This one was meant to be temporary; a way to forcefully kill a misbehaving game without having to reboot the device.

hiulit, 2022-12-02:
I've tested it and it works with <kbd>alt + enter</kbd> but I have a full-size keyboard and it only works with the right most <kbd>enter</kbd>, not the one in the middle. I don't know if the code differentiates between <kbd>enter</kbd> and <kbd>return</kbd>

efornara, 2023-02-21:
The latest templates in previews have the new shortcut environment variable. This should now work:

    export FRT_EXIT_SHORTCUT=shift-enter

For details, see: https://github.com/efornara/frt/commit/00a1f60f8d01ea1920f1e5c1a4a4a1b3370a3cc7

closed 2023-03-04 by efornara