efornara/frt #44: GLXBadContext

adriaciurana, 2022-11-29:
Hello,
First of all, thank you for your project because I find it fascinating to be able to use godot in arm.

I am trying to run **30_3d_material_testers.pck** using a Beaglebone Black with debian.

But for some reason I get this error and I don't know what it could be. I think it's a libGL bug, but I can't find the problem.

```
 ./frt_201_351_arm32v7.bin --main-pack 30_3d_material_testers.pck --verbose
Godot Engine v3.5.1.stable.frt.6fed1ffa3 - https://godotengine.org
X Error of failed request:  GLXBadContext
  Major opcode of failed request:  154 (GLX)
  Minor opcode of failed request:  6 (X_GLXIsDirect)
  Serial number of failed request:  99
  Current serial number in output stream:  98
```
For example if I run: glxgears I get the same error.

Any idea?
Thanks

efornara, 2022-11-30:
> I have found the solution by modifying /etc/lightdm/lightdm.conf and adding xserver-command=X -core +iglx

Does it mean that 30_3d_material_testers.pck works? It's quite heavy and I haven't been testing it for ages. I was even planning to either remove it or reexport it from a more recent version of Godot.

If it does work, does it work well? Can you check that it is using hardware acceleration and not something like llvmpipe (-v/--verbose should tell you).

I would have suggested starting with Dodge the Creeps to be honest.

adriaciurana, 2022-11-30:
Thank you very much for your comment, for now I have not been able to execute it as you say, it is very heavy.

I am trying to install the drivers for the hardware acceleration but it is being very complicated because there is practically no information and it is obsolete, for now I have only managed to execute it via CPU.

efornara, 2022-12-01:
I know what you mean. Unless you cannot boot from MicroSD and you have to flash the device (the risk of bricking it might not be worth it), it might be easier to use an old image with hardware acceleration already included.

Note that the templates I publish on sourceforge require at least debian 10. There is one that should work on debian 9 in previews (frt_201_345_arm32v7_llvm_stretch).

efornara, 2022-12-01:
The error "No available video device" is not Godot or FRT specific and comes straight from SDL2.

Some ideas:

- Set a device-specific SDL2 environment variable before starting the executable
- Compile an older version of SDL2.
- Install a device-specific SDL2 that someone patched and prebuilt.
 
But on the forum you linked they would know better than I do.

adriaciurana, 2022-12-01:
Thank you very much, I think I've managed to come up with something that will work :D Although I'm still on an old 9 debian.
For some reason using:
```
debian@beaglebone:~/game$ ./frt_201_345_arm32v7_llvm_stretch.bin --main-pack 32_3d_dodge_the_creeps.zip
Godot Engine v3.4.5.stable.custom_build.f9ac000d5 - https://godotengine.org
loaded module : gbm_pvr.so
found valid GBM backend : gbm_pvr.so
OpenGL ES 2.0 Renderer: PowerVR SGX 530
OpenGL ES Batching: ON
 
Illegal instruction

```
I remember that when I tried with the CPU version, I had to use v6 instead of v7 because I got the same error. Do you know if a v6 is available to see if it works?

On the other hand, I am going to try to discover what causes this Illegal instruction.

Thank you very much.

efornara, 2022-12-01:
The CPU probably cannot handle the fpu option I'm using (-mfpu=neon-vfpv4). I don't have an environment to compile a v6 stretch, but it shouldn't be needed. I see if I can compile a v7 with a less demanding fpu option (-mfpu=vfp) and upload it on sourceforge.

adriaciurana, 2022-12-01:
Thank you very much! If it is possible to access **frt_201_345_arm32v7_llvm_stretch.bin** and **frt_201_351_arm32v7.bin** with that set of instructions that would be amazing!

I'll try to make a tutorial when everything is working, I think it can be very valuable for future developers to be able to run BBB with Godot and hardware acceleration. 

At least for me it was for the reason that I decided to use it (apart from the fact that rp4 is sold out everywhere).

adriaciurana, 2022-12-05:
I have tried to use the new image, and it seems that it is not enough ;(
I get a segmentation fault and I am not very clear where it is occurring.

````
debian@beaglebone:~/game$ DISPLAY=:0 ./frt_201_345_arm32v7_llvm_stretch_vfp.bin --main-pack 32_3d_dodge_the_creeps.zip
Godot Engine v3.4.5.stable.custom_build.f9ac000d5 - https://godotengine.org
xcb_connection_has_error() returned true
loaded module : gbm_pvr.so
found valid GBM backend : gbm_pvr.so
OpenGL ES 2.0 Renderer: PowerVR SGX 530
OpenGL ES Batching: ON
ALSA lib confmisc.c:767:(parse_card) cannot find card '0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_card_driver returned error: No such file or directory
ALSA lib confmisc.c:392:(snd_func_concat) error evaluating strings
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_concat returned error: No such file or directory
ALSA lib confmisc.c:1246:(snd_func_refer) error evaluating name
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM default
WARNING: All audio drivers failed, falling back to the dummy driver.
     at: initialize (servers/audio_server.cpp:222)

WARNING: MSAA must be <= GL_MAX_SAMPLES, falling-back to GL_MAX_SAMPLES = 4
     at: _render_target_allocate (drivers/gles2/rasterizer_storage_gles2.cpp:5108)
Segmentation fault

````

efornara, 2022-12-05:
Yes, it's hard to tell where the problem is. I'll have to recheck if FRT can handle running with the dummy audio driver. It is also worth trying the 2d platformer instead of a 3d game.

It shouldn't be a problem, but I wasn't expecting you to use DISPLAY=:0. From the thread you linked, it seemed that hardware-acceleration was not available under X11.

efornara, 2022-12-06:
I've uploaded a new file ( frt_201_345_arm32v7_llvm_stretch_vfp_sig.zip ) in previews. It isn't a full debug build, but it isn't stripped of symbols and it should print some sort of backtrace on segfault.

```diff
diff --git a/detect_godot3.py b/detect_godot3.py
index 4e0c07c..661dbab 100644
--- a/detect_godot3.py
+++ b/detect_godot3.py
@@ -52,7 +52,7 @@ def configure_arch(env):
 		env.Append(CCFLAGS=['-march=armv6', '-mfpu=vfp', '-mfloat-abi=hard'])
 		env.extra_suffix += '.arm32v6'
 	elif env['frt_arch'] == 'arm32v7':
-		env.Append(CCFLAGS=['-march=armv7-a', '-mfpu=neon-vfpv4', '-mfloat-abi=hard'])
+		env.Append(CCFLAGS=['-march=armv7-a', '-mfpu=vfp', '-mfloat-abi=hard'])
 		env.extra_suffix += '.arm32v7'
 	elif env['frt_arch'] == 'arm64v8':
 		env.Append(CCFLAGS=['-march=armv8-a'])
@@ -79,7 +79,7 @@ def configure_cross(env):
 
 def configure_target(env):
 	if env['target'] == 'release':
-		env.Append(CCFLAGS=['-O2', '-ffast-math', '-fomit-frame-pointer'])
+		env.Append(CCFLAGS=['-ffast-math'])
 	elif env['target'] == 'release_debug':
 		env.Append(CCFLAGS=['-O2', '-ffast-math'])
 	elif env['target'] == 'debug':
diff --git a/frt_godot3.cc b/frt_godot3.cc
index 5f61284..110e6db 100644
--- a/frt_godot3.cc
+++ b/frt_godot3.cc
@@ -402,7 +402,21 @@ public: // EventHandler
 
 #include "frt_lib.h"
 
+#include <execinfo.h>
+#include <signal.h>
+#include <stdlib.h>
+
+static void signal_handler(int sig) {
+	void *buf[100];
+	int n = backtrace(buf, 100);
+	backtrace_symbols_fd(buf, n, 2);
+	exit(1);
+}
+
 extern "C" int frt_godot_main(int argc, char *argv[]) {
+	signal(SIGSEGV, signal_handler);
+	signal(SIGFPE, signal_handler);
+	signal(SIGILL, signal_handler);
 	frt::Godot3_OS os;
 	Error err = Main::setup(argv[0], argc - 1, &argv[1]);
 	if (err != OK)
```

efornara, 2023-02-21:
I don't know if you are still interested. If you still are, in the previews directory of sourceforge there are packages for FRT 2.1.0 beta. They go as far as I can to support Beaglebone Black. The 3.5.x zip files include arm32v6 versions and are compiled in debian stretch.

closed 2023-03-03 by efornara