efornara/frt #45: FRT 2.1 ToDo List

efornara, 2023-02-02:
Medium Priority

- [x] Godot 4 initial support
- [x] Icon
- [x] can_draw
- [x] New docker images (stretch+clang14)
- [x] arm32v6 chroot+qemu
- [ ] Exit shotcut
- [ ] Web site
- [ ] Some more documentation
- [ ] Audio supporting more than 2 channels
- [ ] Joystick vibra
- [ ] Standard cursor shapes
- [ ] Custom cursor shapes
- [ ] Window events (enter/leave, focus)
- [ ] Keyboard policies
- [ ] Copy and paste

Low Priority

- [ ] Common detect*.py funcs / classes
- [ ] Batch compile on every stable release
- [ ] Find breaking points (correct FRT_GODOT_VERSION in ifdefs)

Cebion, 2023-03-31:
Hi efornara,

I'm doing some testing on my RK3326 SOC RG351v on different CFWs.
Some time ago we started a project called Portmaster to bring various games to the RK3326, RK3356 Platforms with no X11/ Wayland and supplying all tools to handle the controls.

So for each device we detect the correct SDLGAMECONTROLLERDB entry and do a 
export SDL_GAMECONTROLLERCONFIG before running the program.

For games that don't support the SDL2 Joystick detection we use a tool called GPTOKEYB to translate the SDL2 Joystick to either an Xbox360 Pad or to simulate Mouse and keyboard. 

I started using an 2.16 game (Tanks of Freedom)
The detection works but my Analog stick is inverted in the campaign.
This behaviour was also reported here: 
https://old.reddit.com/r/godot/comments/rnoi0v/working_on_a_3d_platformer_for_the_rg351_handheld/ and also here https://github.com/efornara/frt/issues/37

Also interesting because he changes it in the Godot project settings. So maybe there is a default value off? 

I have no idea why but on another OS ArkOS and uOS the DPAD is switched and the Analogue stick is fine,

I encountered games where the SDL Detection is kind of off where I simply use GPTOKEYB then to just map Keyboard controls to my device buttons. But this only works if I can disbale the internal Joysticl even if it its through disabling the SDL_Joystick code.

So my question is if you have any idea what might cause this issue and if it is possible to disable the joystick detection via an argument. 

Similar to FRT_NO_EXIT_SHORTCUTS=FRT_NO_EXIT_SHORTCUTS which I have to set so the game is not exited when pressing start.

I saw on pumkinwaffles git that he exports a custom sdl config before starting his frt
https://github.com/Pumpkinwaffle/EmuELEC-OGS-Godot

But this seems really hacky and the default that works every should also work. It's a standard after all :P

efornara, 2023-03-31:
@Cebion 

Sorry, but I cannot quite understand when you are talking about games in general (e.g. a plain C++ game) and when Godot/FRT games specifically.

#37 was about FRT 1. Feedback for FRT 2 (#39) was different. I assume you are using FRT 2.1.0.

> So for each device we detect the correct SDLGAMECONTROLLERDB entry and do a
export SDL_GAMECONTROLLERCONFIG before running the program.

This is essentially what pumkinwaffles is doing, but your remark about it means that I should probably ask if you are exporting SDL_GAMECONTROLLERCONFIG before running Tanks of Freedom too.

Cebion, 2023-03-31:
Sorry for the wall of text :P
Until the part of Tanks of Freedom I was talking about general c++ games using SDL2 to show you my train of thoughts.

The rest is all about testing this specific games on various OSes with FRT2.1.0

Yes, I'm talking about FRT 2.1.0 compiled freshly yesterday.

Yes, I'm also exporting SDL_GAMECONTROLLERCONFIG before running FRT2.

My os also sets SDL_GAMECONTROLLERCONFIG_FILE=/storage/.config/SDL-GameControllerDB/gamecontrollerdb.txt
with my device specific config in the ENV at boot.
So the controls still get found and mapped like with manually setting the SDL_GAMECONTROLLERCONFIG.

I just tried not loading SDL_GAMECONTROLLERCONFIG at all  and also removed all ENV SDL_GAMECONTROLLER settings and the game still has controls with the analog stick disabled.

I'm starting the game via 

./frt_200_216_arm64v8.bin -main_pack ./data.pck

Maybe it ignores the file alltogether :D

efornara, 2023-03-31:
Here is how FRT works.

SDL2 exposes two APIs: a low level one (Joystick) and a high level one (GameController). Unlike some SDL2 games, FRT uses the low level API. It delegates controller mapping to Godot. Like SDL2, Godot uses SDL_GAMECONTROLLERCONFIG to extend its own internal mapping db. Unlike SDL2, it ignores SDL_GAMECONTROLLERCONFIG_FILE (AFAIK).

One thing that might cause issues with Tanks of Freedom is that the Godot 2 FRT export template has a non-empty internal mapping db. In Godot 3/4, it is empty. Maybe, if a mapping for a given GUID already exists in its db, Godot 2 ignores the SDL_GAMECONTROLLERCONFIG override and keeps using the internal one.

By any chance, have you ever tried a Godot 3 game to see if mapping works there?

Cebion, 2023-03-31:
Thank you for the explanation! 

Okay I just tested with a Godot 3 game (crossing-the-sinai) 
It's a Godot 3.0.6 game.
I unset all my SDL_GAMECONTROLLERCONFIGs started the game with a keyboard connected and then tested the controls
And indeed as before Dpad works as expected and Analogue Sticks are inverted.

For the next test I tried to start it with 

export SDL_GAMECONTROLLERCONFIG="03000000091200000031000011010000,OpenSimHardware OSH PB Controller,a:b0,b:b1,y:b3,x:b2,start:b6,back:b7,dpup:h0.1,dpleft:h0.8,dpdown:h0.4,dpright:h0.2,leftshoulder:b4,lefttrigger:b10,rightshoulder:b5,righttrigger:b11,leftstick:b8,guide:b9,leftx:a0~,lefty:a1~,rightx:a2,righty:a3,platform:Linux,"
./frt_210_306v1_arm64v8.bin --main-pack ./sinai_linux.pck

// No difference between the export and no export.

To test if its used at all I switched the dpad values around and tried again. No change dpad still works as intended.

Now I saw that on these devices I'm working with have weird EVTEST mappings.

On evtest I press for example L2 and R2 and I get 
```
Event: time 1680288997.864968, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000b
Event: time 1680288997.864968, type 1 (EV_KEY), code 314 (BTN_SELECT), value 1
Event: time 1680288997.864968, -------------- SYN_REPORT ------------
Event: time 1680288997.969965, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000b
Event: time 1680288997.969965, type 1 (EV_KEY), code 314 (BTN_SELECT), value 0
Event: time 1680288997.969965, -------------- SYN_REPORT ------------
Event: time 1680288998.949929, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000c
Event: time 1680288998.949929, type 1 (EV_KEY), code 315 (BTN_START), value 1
Event: time 1680288998.949929, -------------- SYN_REPORT ------------
Event: time 1680288999.089909, type 4 (EV_MSC), code 4 (MSC_SCAN), value 9000c
Event: time 1680288999.089909, type 1 (EV_KEY), code 315 (BTN_START), value 0
Event: time 1680288999.089909, -------------- SYN_REPORT ------------
```
So I guess used the correct values for our SDL2 config file but the low level joystick has wrong values.
Could it be that it looks for these values that are exposed to evtest?

Thanks in advance!

efornara, 2023-04-01:
It turns out that even Godot 3.0.6 has a non-empty db. The change happened sometime later. It doesn't matter, though. The GUID you are using (03000000091200000031000011010000) doesn't look like it is in its list, so no conflict there.

Regardless of how the "wrong" values get into the engine from the lower layers (libinput, SDL2 or even FRT), you should be able to remap them using SDL_GAMECONTROLLERCONFIG. That's the theory. Looking at the code, Godot 3.0.6 (and 2.1.6) seems to ignore the "~" suffix:

https://github.com/godotengine/godot/blob/3.0.6-stable/main/input_default.cpp#L1226-L1230

Sometime later, upstream started handling it. Here is Godot 3.5.2:

https://github.com/godotengine/godot/blob/3.5.2-stable/main/input_default.cpp#L1178-L1182 

So, leaving aside the DPad (which should be fixable), it looks like there is no easy way to invert the analog axis in the oldest versions of Godot.

Cebion, 2023-04-01:
Thanks for the research! I will handle it for now to make a fork and try to disable all joysticks all together and will map the keyboard /mouse controls to the gamepad with gptoleyb.

Cebion, 2023-04-13:
Hi efornara,

today I encountered a really weird issue.
I worked with the project https://gitlab.com/team-potato/titan_tactics/

I opened the project in Godot 3.2.3 which is the version of the project.
Exported the .pck and it intially runs but gets stuck at selecting the map.

I got following error:

```
ERROR: _load: Condition "found" is true. Returned: RES()
   At: core/io/resource_loader.cpp:279.
ERROR: poll: res://scenes/gui.tscn:11 - Parse Error: [ext_resource] referenced nonexistent resource at: res://gfx/UI.png
   At: scene/resources/resource_format_text.cpp:440.
ERROR: load: Condition "err != OK" is true. Returned: RES()
   At: core/io/resource_loader.cpp:208.
ERROR: _load: Condition "found" is true. Returned: RES()
   At: core/io/resource_loader.cpp:279.
ERROR: poll: res://scenes/world.tscn:8 - Parse Error: [ext_resource] referenced nonexistent resource at: res://scenes/gui.tscn
   At: scene/resources/resource_format_text.cpp:440.
ERROR: load: Condition "err != OK" is true. Returned: RES()
   At: core/io/resource_loader.cpp:208.
ERROR: _load: Condition "found" is true. Returned: RES()
   At: core/io/resource_loader.cpp:279.
```
All files are in the correct location and when running the game in the editor all works well.
I researched the error and tried all kind of things like reimporting all files but could not solve the error.

In the end I stumbled upon the hint to manually reimport on of the assets the UI.png there it asks to Import as what file and by default most files are imported as type "Texture". 
Just to test things I imported the UI.png file as Type "Image" instead and tried again.

This time it worked.

It still gave me some errors but it loaded and worked!

```ERROR: _load_data: Condition "!f" is true. Returned: ERR_CANT_OPEN
   At: scene/resources/texture.cpp:502.
[[Resource:1537]]
[World] Loading level level_1_3d
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
ERROR: get_parent_class: Condition "!ti" is true. Returned: StringName()
   At: core/class_db.cpp:343.
spawn entered tree
spawn entered tree
spawn entered tree
spawn entered tree
{added_tiles:151, connections:218, excluded_tiles:0, removed_tiles:0}
astar pathfinding populated
spawn ready
spawn ready
spawn ready
spawn ready
```

Is this a problem on the godot / project side or a thing that needs to be adressed on the FRT side?

Thanks in advance!

Cebion, 2023-04-13:
> 

Thanks thought of something like that.

Cebion, 2023-04-21:
Another hopefully known error.

Tried to compile Godot 3.0.6 today from https://downloads.tuxfamily.org/godotengine/3.1/godot-3.1-stable.tar.xz

Compiled via scons platform=frt tools=no target=release use_llvm=yes module_webm_enabled=no

`root@DE-NB-06622:~/godot/godot-3.0.6-stable# scons platform=frt tools=no target=release use_llvm=yes module_webm_enabled=no -j8
scons: Reading SConscript files ...
Checking for C header file mntent.h... (cached) yes
scons: done reading SConscript files.
scons: Building targets ...
[Initial build] Compiling ==> platform/frt/frt_godot3.cc
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btCompoundShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btConcaveShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btConeShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btConvexHullShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btConvexInternalShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btConvexPointCloudShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btConvexPolyhedron.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btConvexShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btConvex2dShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btConvexTriangleMeshShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btCylinderShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btEmptyShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btHeightfieldTerrainShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btMinkowskiSumShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btMultimaterialTriangleMeshShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btMultiSphereShape.cpp
platform/frt/frt_godot3.cc:13:10: fatal error: 'drivers/gles2/rasterizer_gles2.h' file not found
#include "drivers/gles2/rasterizer_gles2.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btOptimizedBvh.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btPolyhedralConvexShape.cpp
[Initial build] Compiling ==> thirdparty/bullet/BulletCollision/CollisionShapes/btScaledBvhTriangleMeshShape.cpp
1 error generated.
scons: *** [platform/frt/frt_godot3.frt.opt.llvm.o] Error 1
scons: building terminated because of errors.`

Seems 3.0.6 doesn't have the GLES2 folder.
3.1 does have it but copying it over results in a slew of other errors like 

`./drivers/gles2/rasterizer_storage_gles2.h:34:10: fatal error: 'core/pool_vector.h' file not found
#include "core/pool_vector.h"`

So not really compatible.

Now I've seen that your 3.0.6 version is called V1 is there a reason for that?

efornara, 2023-04-21:
For Godot 3.0.6, FRT uses a version with a limited version of the GLES2 renderer backported. It is good enough to run basic 2D games. You can find it here: https://github.com/efornara/godot/releases/tag/3.0.6-gles2-v1 (branch https://github.com/efornara/godot/tree/3.0-gles2)

It should be possible to fix FRT to make it compile with upstream 3.0.6, but I don't see it as a high priority task since you can still access the unmodified 3.0.6 GLES3 renderer from 3.0.6-gles2-v1 by adding "--video-driver GLES3".

hiulit, 2023-05-15:
Hi there!
Is there a way to rotate the screen in FRT 2.x upon launch? Some paramater maybe?
Thank you!

efornara, 2023-05-17:
> Is there a way to rotate the screen in FRT 2.x upon launch?

Not that I'm aware of. It would be SDL2-specific, and I don't think there is a generic environment variable for that. Your platform might have a patched SDL2 that uses a proprietary library to "fix" the issue. IIRC, some Odroid Go Advance distros had something like this. If not, your best option is likely to change the camera settings in your game.

hiulit, 2023-05-17:
Thanks for the anwer! It's not an issue :D
I deliberately would like to be able to rotate the whole window before starting the game and not mess with the internal code :P
Would it be possible to add such a thing to FRT? An SDL2 envinronment variable or something like that?

efornara, 2023-05-17:
> Would it be possible to add such a thing to FRT?

No. The mechanism used to create the display / graphics context is hidden inside SDL2. 

> An SDL2 envinronment variable or something like that?

SDL2 environment variables are used by SDL2 transparently. For example, you can select Wayland vs X11 by setting an SDL2 environment variable without even telling FRT. If a (custom) version of SDL2 supports rotation, I guess they might do so automatically or choose to expose a non-standard environment variable to enable it.