efornara/frt #46: Banana pi m1+ segfault

tccolin70, 2023-04-21:
Hi,

I am trying to run frt on my banana pi m1+ (armv7 with neon and vfpv4 and a mali 400 MP2 gpu) on a homemade lightweight distro but it segfaults
I don't think it is an issue with my libraries since other programs work fine (ioquake3, glmark2-es2)

I have tried running both 30_3d_material_testers.pck (with "--video-driver GLES2" because mali400 doesn't support GLESv3) and 32_3d_dodge_the_creeps.zip with both the armv6 and armv7 binaries but they all segfault at the same point

i can see the godot logo on my screen but it segfaults immediately after

```
bash-5.2# ./frt_210_352_arm32v6.bin --main-pack 32_3d_dodge_the_creeps.zip -v

Godot Engine v3.5.2.stable.frt.170ba337a - https://godotengine.org
WARNING: Started the engine as `root`/superuser. This is a security risk, and subsystems like audio may not work correctly.
Set the environment variable `GODOT_SILENCE_ROOT_WARNING` to 1 to silence this warning.
     at: setup2 (main/main.cpp:1368)
sh: setterm: not found
Using GLES2 video driver
OpenGL ES 2.0 Renderer: Mali-400 MP
OmniLight cubemap shadows are not supported by this GPU. Falling back to dual paraboloid shadows for all omni lights (faster but less.
OpenGL ES 2D Batching: ON
Batching Options:
        max_join_item_commands 16
        colored_vertex_format_threshold 0.25
        batch_buffer_size 16384
        light_scissor_area_threshold 1
        item_reordering_lookahead 4
        light_max_join_items 32
        single_rect_fallback False
        debug_flash False
        diagnose_frame False
 
Loading resource: res://default_bus_layout.tres
CORE API HASH: 0
EDITOR API HASH: 0
WARNING: MSAA must be <= GL_MAX_SAMPLES, falling-back to GL_MAX_SAMPLES = 4
     at: _render_target_allocate (drivers/gles2/rasterizer_storage_gles2.cpp:5130)
Segmentation fault
bash-5.2#
```
Any ideas?

efornara, 2023-04-21:
Realistically, on a Mali 400 you can only expect 2D from a Godot 3 project.

You _might_ be able to get 3D Dodge The Creeps to run somehow on a Mali 400, but you might need to reexport it from the original https://github.com/godotengine/godot-3d-dodge-the-creeps. Beside converting the project to GLES2, you will also need to reduce the shadow map size. In theory, these changes are already in the PCK you downloaded. In practice, later versions of FRT look for the shadow map size in a different place. I see if I can fix 3D Dodge The Creeps demo one of these days, but keep in mind that, IIRC, on my H3 (Mali 400) I had some rendering artifacts.

As for the 3.0 3D Material Tester demo, I don't expect it to run on a Mali 400 ever. it probably makes sense for me to delete it for good.

tccolin70, 2023-04-21:
I'm aware of the performance yes, I'm only going to run simple projects on this board but my issue here is that it segfaults before rendering even a single frame

tccolin70, 2023-04-23:
Hi, 
Sorry for the delay, i had to port malifb patches to a more recent version of SDL to get frt to work, turns out there was an issue with my libraries but the godot player crashed too early with the 3d example to know haha

I'm happy to say that you were right about godot's 3d mode being broken on mali 400, the 3d examples still segfault but the 2d platformer example runs well!

I will continue to debug the 3d mode on my side to see where it goes wrong exactly and to see if maybe i could fix it, but this is an issue with godot itself not frt so for now i will mark this as completed. 
There is also an issue with input not working but this is likely because of my kernel or SDL. I will open a new issue if it turns out to be an issue with frt.

Thanks for the help and for making this awesome project!